MLB-LED-Scoreboard/mlb-led-scoreboard #6: Create Main Menu

ajbowler, 2018-01-28:
This would be the new landing page when you spin up the board. Should have a nice graphic and a menu to navigate to either MLB Standings (see #4) or a list of games to pick from (see #7).

Because this requires a sexy graphic, I'm not gonna do it cause I lack the confidence.

ajbowler, 2018-02-03:
Putting on hold because this will require some keyboard input, see #7

craigm26, 2018-02-16:
are you planning to have a GUI via HDMI for this? I could do a small python or js server to help control the scoreboard. It'd be nice to control with a smart phone via web browser

ajbowler, 2018-02-16:
No, ideally there wouldn't be any extra GUI for this, trying to keep the hardware aspect of this down as much as possible. The LED board would show a main menu on startup, then the user can use a keyboard plugged into the Pi without any other hardware to navigate.

twfarley, 2018-02-16:
What are you thinking of for a graphic for this? I might be able to help out.

ajbowler, 2018-02-16:
The only thing that I know we _don't_ want is the MLB logo. Legal issues and all that which I'm not interested in getting involved with.

Right now the only criteria I'd like to see is "Games" and "Standings" for the user to pick from, and some kind of title graphic. Tricky part is it all needs to fit on a 32x32 pixel display.

craigm26, 2018-02-16:
Can the display do a transition animation?  menu then select today’s games (or team or standings), plays animation

I like the idea of keyboard only operation. Use arrow keys and maybe a shutdown command will make a huge difference.

ajbowler, 2018-02-16:
@craigm26 That's a neat idea to add after the fact, I'll log a separate issue for it after this one gains some traction.

ajbowler, 2018-02-16:
@craigm26 @twfarley @swemoney I've created a Slack workspace for any general discussions to keep the issues a little cleaner. https://join.slack.com/t/mlb-led-scoreboard/shared_invite/enQtMzE2NjI0MTY3MTI0LTAyZjhlZjk4NWEwZGUyNGM4NGQ4MDM1MDA0NzdhODU0MzRjM2EyMjk2ZDc3YTkyZmI3NDA5N2VkMzExZDQxYmQ

craigm26, 2018-02-16:
[image]

ajbowler, 2018-04-12:
We've committed to the config file route at this point so having another way to do things will be pretty confusing. Closing this issue and other keyboard related ones, at least for now.

closed 2018-04-12 by ajbowler

bryantee, 2019-12-24:
I think a way to configure via the web sounds like the ideal solution.

twfarley, 2019-12-24:
#138 should cover that.