MLB-LED-Scoreboard/mlb-led-scoreboard #8: Display Final with winning and losing pitchers when game is over

ajbowler, 2018-01-28:
Currently it shows the last at-bat when a game is over, instead it would be nice to show something like "Final" with the winning and losing pitchers below the scores.

ajbowler, 2018-02-03:
The catch - this will have to be scrolling text because the names will be too long. The [base RGBMatrix library](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/runtext.py) has an example for running text across the screen that could be similarly done here. The text will need to be shifted a position every refresh (something fast, maybe `time.sleep(0.25)`)

Some good text to see scrolling across:

Winner: Lester (10-2) Loser: Wainwright (0-99)

swemoney, 2018-02-09:
Can we hard code "Wainwright (0-99)" for every gameover?

I'd like this idea. It could go well with 64x32 support (more room for the names) depending on what is decided would go good on that side of the screen after a game is over.

swemoney, 2018-02-10:
I wonder if we want to break this down into a couple of different issues? We're going to need to cache the gamedata and only actually hit the servers every 15 seconds since scrolling the data requires us to reduce the sleep time to .25 or so. The caching might be a pretty decent overhaul in itself.

ajbowler, 2018-02-10:
It won't require another server hit. In the loop we're already refreshing every 15s, we just add an inner loop that scrolls the pitcher text with a different sleep. Before we render the screen we grab all the game_event data and can check if it's reached Final. If it has, render the team banners and write an inner loop for the pitcher text, otherwise just render  the scoreboard without the other loop.

swemoney, 2018-02-10:
Could you steer me in the right direct for how to render just the text on screen? I actually had the whole separate loop working but when I went to do the rendering I wasn't sure how to write to the screen without it flashing everything. I'll give it another look because that was my initial solution.

ajbowler, 2018-02-10:
So I would prefer to render both the team banners and then instead of the pitches/outs/bases, scroll the pitcher text across below the banners.

Had to do some digging in mlbgame's API to see what we can use.

We currently get the game overview [here](https://github.com/ajbowler/mlb-led-scoreboard/blob/master/scoreboard.py#L20). That API returns an object that contains the abbrevs we need for the banner which is why we use it. It also contains a `status` field that will say "Final" when the game's over.

Below that we execute a try block to get all the other goodies from the `game_events` API like the bases, outs, pitches, and inning. Instead of doing this, we can do a check on the status. If it's over, just return the status, and wherever we render the scoreboard, check our `scoreboard` object [here](https://github.com/ajbowler/mlb-led-scoreboard/blob/master/scoreboard_renderer.py#L15) if we have a Final status instead of inning information. If we do, execute an inner loop that draws the board every .25 or whatever we decide to set the sleep to.

ajbowler, 2018-02-21:
After the recent refactoring of the game renderer I think this became easier. We just need to set the refresh rate to the same as the pregame rate if the game status is FINAL, then we can scroll the winning/losing pitcher text across like we do for pregames. The current loop will still rotate after 15 seconds (this will change depending on the scrolling text length, see #30 )

ajbowler, 2018-02-28:
Closed by https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/commit/dc809f48f9141374674c9d0abba038f746bec86a

closed 2018-02-28 by ajbowler