MLB-LED-Scoreboard/mlb-led-scoreboard #9: 64x32 enhancements

ajbowler, 2018-01-29:
Logging this issue to see if there's any interest. I'll keep it open throughout the season and if nothing happens I'll close this, otherwise, anybody interested in this project can throw out some ideas.

Current ideas:

Display pitcher vs. batter info on the second board, including ERA, avg/obp/ops, etc.
Display flashing marquees similar to MLB's GameDay, such as "HOME RUN!"

swemoney, 2018-02-09:
I'm definitely interested in 64x32 support. Both of those ideas are exactly what I thought would go great there. Pitcher v Batter and then replace with a short marquee when things happen (strikeout, homerun, etc).

ajbowler, 2018-02-10:
I'll add an extra label for 64x32 related items, it seems we have the interest for it

closed 2018-02-10 by ajbowler