MLB-LED-Scoreboard/mlb-led-scoreboard #21: Don't rotate if passed team is currently playing

swemoney, 2018-02-09:
A suggestion for the current rotate feature would be to check if a team was also passed and stop rotating while that team is playing and continue shortly after their game ends.

ajbowler, 2018-02-09:
So the current behavior is to not rotate at all if you pass a team and they have a game. It will only rotate through games if you pass the `-r` flag. Are you just proposing to enable that flag when the passed team's game ends if it isn't already?

swemoney, 2018-02-09:
I'm proposing if the -r flag is set and a team is also set, then rotate today's games until the passed team's game is on (so focus on the passed team's game) and then go back to rotating maybe a half hour after the game ends or something.

ajbowler, 2018-02-09:
Currently it goes straight to the game of the team you requested rather than rotating over towards it. I'd rather keep that behavior.

I do like the idea of starting up the rotation after the passed team's game is over. Half an hour sounds like a good window before starting.

Can't really test this well until the season starts, but I'll add this to the list.

swemoney, 2018-02-09:
The benefit of also rotating before a game has started comes when the provided team has a late game, your scoreboard can rotate through other games during the rest of the day and then stop rotating once the "preferred team" is ready to play. A half hour before the game starts (feels like a good amount of time for this too) maybe the rotation can stop and you'll have a "game preview" of sorts and the scoreboard can remain on that game until it's over.

ajbowler, 2018-02-09:
I'd be fine with this as long as it's explicitly documented in the README.

The conflicting workflow is wanting to know when my team's game starts as soon as I turn it on. If it turns out to be a night game I won't know until it rotates all the way over to it.

swemoney, 2018-02-09:
Right. That makes the most sense. I agree that it should still start showing the selected team right away (regardless of when that game starts) before it starts rotating through other games. Maybe even adding a separate option that makes the rotation more of a "screensaver". Where after a certain amount of time that the selected team isn't playing, it starts rotating through other games until that game is going to start soon?

ajbowler, 2018-02-09:
To simplify this, the `-r` flag should rotate every 15 seconds if a team is not passed.

If a team is passed and it has a game that day, the board will show either the game in progress, or the "starting game" display (just logged #22).

If the game has started, the rotation will pause until 30 minutes after Final.
If the game has not started, the rotation will continue after 15 seconds as normal.
If the game is starting within 30 minutes, the rotation will pause.

Sound good?

swemoney, 2018-02-09:
That sounds perfect!

ajbowler, 2018-02-09:
This is something that will have to be tackled after the season starts because right now everything just locks at the 9th inning (unless #8 is tackled). If you want to do this let me know, otherwise I'll work on it at some point.

swemoney, 2018-02-10:
I'm taking a look at it now. Still pretty new to rendering on the LED matrix but I'll see what I can do.

ajbowler, 2018-02-11:
I added a change that will help pave the way for #22 and any other issues that require the board to keep track of how much real time has passed (basically an outer loop).

I'm gonna try 15 minutes instead of 30 and see how it goes. If you or other users prefer a different full-refresh time, that might be a candidate for another CLI argument.

ajbowler, 2018-02-26:
Updating this issue since we've switched over from command line args to a config file and are pulling from a much friendlier API from mlbgame.

- [ ] Add config option `disable_rotate_for_preferred_team`
- [ ] Update README for the new flag
- [ ] Add logic to the game render loop to, if the flag is enabled, not rotate if the current game is the preferred team's and that game's status is marked as "Warmup" or "In Progress".

ajbowler, 2018-03-02:
Adding onto this, I'm gonna put the idea of having a time window before warmup and after final on the back burner and just follow the checklist above, should keep the scope of the commit for this change easy to figure out.

PR #76 (2018-03-09): Keep scoreboard locked on the preferred team when that game is live
Add a config option (default is true) to keep the scoreboard pinned to your preferred team while that team is currently in warmup, live or in "game over" status (which seems to last for about 5 minutes after the game ends).

Closes #21 (We can create a new issue to allow us to stay on the final scoreboard for a set amount of time)
closed 2018-03-09 by ajbowler