MLB-LED-Scoreboard/mlb-led-scoreboard #27: Launch script from a different directory?

twfarley, 2018-02-16:
I was trying to add the python script to launch on boot but it doesn't work unless I'm in the mlb-led directory. For example:

$ sudo python ~/mlb-led-scoreboard/main.py -r

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 33, in <module>
    render_games(matrix, canvas, games[0], args)
  File "/home/pi/mlb-led-scoreboard/renderer.py", line 30, in render_games
    success = __refresh_scoreboard(canvas, game)
  File "/home/pi/mlb-led-scoreboard/renderer.py", line 89, in __refresh_scoreboard
    renderer = ScoreboardRenderer(canvas, scoreboard)
  File "/home/pi/mlb-led-scoreboard/scoreboard_renderer.py", line 9, in __init__
    self.colors = json.load(open('Assets/colors.json'))
IOError: [Errno 2] No such file or directory: 'Assets/colors.json'

ajbowler, 2018-02-16:
I've got an idea in mind to get this working.

closed 2018-02-17 by ajbowler

twfarley, 2018-02-17:
Awesome! Thanks! I’ll test this over the weekend.

commit b75bbabd (2018-02-17): Allow running the project from any working directory

Closes #27