MLB-LED-Scoreboard/mlb-led-scoreboard #40: Center pregame matchup

ajbowler, 2018-02-23:
Center (as best we can, even numbers and all that) the pregame matchup text depending on matrix width.

[image]

@swemoney has a pretty sweet setup that could use _just a little_ love here.

swemoney, 2018-02-23:
I doubt the pitcher info will ever be short enough to center. My longer screen just starts scrolling from the edge.

ajbowler, 2018-02-23:
Yeah I'm talking the teams and the game time

flemingjw, 2018-02-28:
On the new stretched mode the matchup is centered, but the starters still scroll from the edge

ajbowler, 2018-02-28:
That is expected, any scrolling text will start from the far right edge and go all the way until the end.

(Well, not yet, they will stop after the refresh, see #30 )

commit a2677876 (2018-03-01): Stretch the scoreboard to fit 64x32 matrixes (#67)

* Stretch the scoreboard to fit 64x32 matrixes

Closes #63 
Closes #40
closed 2018-03-01 by ajbowler