MLB-LED-Scoreboard/mlb-led-scoreboard #52: Add option for end of day

swemoney, 2018-02-25:
Right now, we get the current day based on the current time. An option that could be set to a time before switching to the next day so the scoreboard, after midnight, still rotates through scores from the previous day.

ajbowler, 2018-02-25:
Should be done after #29 as that will make things easier here

swemoney, 2018-03-05:
For the time being, I recommend a simple cron job that kills the scoreboard and restarts it at any time you want to roll over to the next day's scores.

swemoney, 2018-03-06:
Better yet, I've added 2 pages to the [wiki](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/wiki) for better controlling your scoreboard and accomplishing exactly this.

PR #98 (2018-03-28): Add config option to change the end of day time
Closes #52 

I chose midnight as the default because that is how the scoreboard operates currently.
closed 2018-03-28 by ajbowler