MLB-LED-Scoreboard/mlb-led-scoreboard #53: Scoreboard crashes when switching to next day

swemoney, 2018-02-25:
The scoreboard does some flickering and eventually crashes to a blank matrix when switching from rotating the games from the day before to displaying the next days games.

ajbowler, 2018-02-26:
Need a little more info here, as I see it from `main.py` it would currently never switch to the next day's games, as we never update the day/month/year values passed to `mlbgame.games`

Rmarken5, 2018-02-27:
I've run into this issue as well. I don't know that it's necessarily try to switch the next day but rather rotate back to the first game. Here's the stack trace that I got. 

> Traceback (most recent call last):
>   File "main.py", line 44, in <module>
>     GameRenderer(matrix, canvas, games[0], config).render()
>   File "/home/pi/mlb-led-scoreboard/renderers/games.py", line 64, in render
>     self.__refresh_game(game, overview.status)
>   File "/home/pi/mlb-led-scoreboard/renderers/games.py", line 97, in __refresh_game
>     scoreboard = Scoreboard(game)
>   File "/home/pi/mlb-led-scoreboard/data/scoreboard.py", line 16, in __init__
>     self.game_data = self.__current_game_data(game)
>   File "/home/pi/mlb-led-scoreboard/data/scoreboard.py", line 27, in __current_game_data
>     game_data['inning'] = self.__current_inning(game_id)
>   File "/home/pi/mlb-led-scoreboard/data/scoreboard.py", line 44, in __current_inning
>     inning_status['at_bat'] = self.__current_at_bat(at_bats[-1])
> IndexError: list index out of range

I did some digging(I'm a beginner when it comes to python), and it looks like it may have something to do with the MLBAPI itself.

swemoney, 2018-02-27:
Give the latest a checkout and see if you're still having issues. A lot of work has been done on the data objects.

Rmarken5, 2018-02-27:
Awesome. That seems to have fixed the issue I was having.

ajbowler, 2018-02-28:
I'm gonna say this issue is fixed by https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/commit/fa191b7850fd69271b1475f762690f14ba57c9c3 and close it, anybody can reopen if they see this again.

closed 2018-02-28 by ajbowler