MLB-LED-Scoreboard/mlb-led-scoreboard #54: Scoreboard starts to flash constantly after 15 minutes

swemoney, 2018-02-25:
When we go to do a full refresh of all the games, it causes the scoreboard to start flashing quickly. This continues until restarting. Only present while displaying pregames, I didn't have this issue while it was rotating through final scores.

5StringThunder, 2018-02-25:
It does this for single active game after 15 minutes too

ajbowler, 2018-02-25:
Currently working to reproduce this today, seeing if it happens when rotate is off.

ajbowler, 2018-02-25:
Able to reproduce on hand, going to add some extra debug logging to track down what's going on. The board is running through the 15s and printing out my preferred team's game to the debug logs, but not rendering anything past the blue backdrop.

I expect whatever is causing @swemoney's original issue is also causing @5StringThunder and my own to have issues with active games.

ajbowler, 2018-02-25:
Fixed. Had the canvas.Fill() in the wrong spot in the loop which caused it to render the board then refill it, which caused the flashing.

Pregames should be fixed as well but re-open if there's still an issue.

commit fa191b78 (2018-02-25): Fix games list refresh after the 15 minute window

Closes #54
closed 2018-02-25 by ajbowler