MLB-LED-Scoreboard/mlb-led-scoreboard #60: 32x64 board - rows flicker

JTskinz, 2018-02-26:
Rows in the board constantly flicker when running the MLB program.   I have not seen the issue running any of the sample programs.

ajbowler, 2018-02-26:
@JTskinz some other users have noticed this as well and have used one of the LED matrix flags, I think it was `--led-slowdown-gpio`. I'll need @swemoney to chime in here cause I haven't done this personally.

Either way, this is a common issue experienced with the LED board bindings and the rgbmatrix maintainers (link in the README) are usually very helpful and active when it comes to hardware issues.

JTskinz, 2018-02-26:
Thanks for the quick reply!  I have tried different values with the --led-slowdown-gpio flag but didn'tfind anything that worked yet.  I will wait and see if @swemoney has any ideas.

ajbowler, 2018-02-26:
Are you using a different power source for your board and your Pi? I've found that it's just much easier to deal with when the board and the Pi are running on their own power sources instead of running just one through the HAT.

I'm by no means a hardware expert so I usually direct any hardware-specific questions to the rgbmatrix maintainers. The scoreboard is simply a user of that project to show some baseball stuffs.

swemoney, 2018-02-26:
Ya. If you are seeing sporadic flickering, --led-slowdown-gpio=2 should clear that up if that's the issue. The issue could also be the 8x2 data cable. The first cable adafruit sent me had one end on backwards and I needed to bump the slowdown up to 3. After I got a new cable, 2 works great. If you're seeing a different kind of flicker, maybe post a video and I can tell if that's what I was getting before.

swemoney, 2018-02-26:
Also, according to the readme on the rpi-rgb-matrix library, there are 2 types of 64x32 displays. One of them requires --led-multiplexing=1. So that's another thing to try. [README](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md)

swemoney, 2018-02-26:
Also, turning off sound on your pi, can help some flickers. And there's a simple PWM mod you can do that's detailed in the "improving flickering" section of the matrix library [README](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/README.md).

ajbowler, 2018-03-02:
@JTskinz were you able to resolve your LED board issues? I'll leave this issue open for another couple days.

JTskinz, 2018-03-02:
@ajbowler yes I was!   Turning off sound and doing the PWM mod mentioned above fixed the issue.

ajbowler, 2018-03-02:
@JTskinz great, glad everything's working!

closed 2018-03-02 by ajbowler