MLB-LED-Scoreboard/mlb-led-scoreboard #61: Add Alexa device support

craigm26, 2018-02-26:
This is based on a hypothetical that I would enjoy in terms of functionality. I currently have Alexa dev accounts and have built skills. It'd be nice to be able to say "Alexa, turn on the scoreboard" and the scoreboard turns on. 

I realize there's a number of challenges associated with this issue, but it's a good opportunity for us to implement this feature. This is low on the totem pole for now.

ajbowler, 2018-02-26:
Basically this sounds like you want an Alexa skill that runs a bash command, cause that's all that's needed to turn on the scoreboard.

"Alexa, turn on the scoreboard" => `sudo python ~/insert_path/mlb-led-scoreboard/main.py`

I think this makes a good candidate for a separate project and possibly combining both this project and the Alexa skill under a GitHub org.

craigm26, 2018-02-26:
Agreed with the direction of the response above. I'll create a github org (non-profit) and see how that goes. The alternate is gitlab as I do enterprise development there on a different pricing scheme. 

I'll report back.

craigm26, 2018-02-26:
[Alex Device repo at new github org](https://github.com/MLB-LED-Scoreboard/alexa-device)

trevor-viljoen, 2018-02-27:
I've also done some Alexa work with AWS Lambdas handling the Alexa intents. Look into https://github.com/Miserlou/Zappa. You would probably need a small webserver running on the Pi, listening for calls from AWS to handle the local functions like turning on the scoreboard or asking for a score update, etc. Perhaps just a flask app, or something from node.js like loopback.io?

craigm26, 2018-02-27:
(flask ask app)[https://github.com/johnwheeler/flask-ask] would be nice to have. 

Would the Alexa part be as a skill or as a product (or both)? I've set up both partially, but I think we really want to simplify it to meet our design needs.

closed 2022-04-13 by twfarley