MLB-LED-Scoreboard/mlb-led-scoreboard #72: Crash during a network issue

swemoney, 2018-03-05:
Leaving the scoreboard running for a while, it looks like it's possible for the Pi to run into some kind of strange network hiccup that completely crashes the scoreboard. We should fail more gracefully on this kind of issue.

`File "/home/pi/mlb-led-scoreboard/main.py", line 42, in <module>
    GameRenderer(matrix, matrix.CreateFrameCanvas(), games, config).render()
  File "/home/pi/mlb-led-scoreboard/renderers/games.py", line 65, in render
    overview = mlbgame.overview(game.game_id)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/__init__.py", line 198, in overview
    return mlbgame.game.Overview(mlbgame.game.overview(game_id))
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/game.py", line 324, in overview
    data = mlbgame.data.get_overview(game_id)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/data.py", line 82, in get_overview
    'linescore.xml'))
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 429, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 447, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1228, in http_open
    return self.do_open(httplib.HTTPConnection, req)
  File "/usr/lib/python2.7/urllib2.py", line 1198, in do_open
    raise URLError(err)
urllib2.URLError: <urlopen error [Errno -3] Temporary failure in name resolution>`

swemoney, 2018-03-08:
This may have been an issue with hitting the API way too much while data was scrolling. Should be fixed in 2462dcd83d307d5b3edea8057915cd846771394b. Going to continue to monitor for now.

ajbowler, 2018-03-13:
Closing, reopen if if the issue pops up again.

closed 2018-03-13 by ajbowler