MLB-LED-Scoreboard/mlb-led-scoreboard #73: Add option to display standings on an off day

alphastanley, 2018-03-05:
Can we embed division standings as an option for off days?  Example move of:

```
standings = mlbgame.standings(datetime.datetime(year, month, day))
division = next(division for division in standings.divisions if division.name == config.preferred_division)
renderers.standings.render(matrix, matrix.CreateFrameCanvas(), division)
```

to the branch logic of:

`if not len(games):`

from main.py.

ajbowler, 2018-03-05:
This would make a good config option.

- [ ] Add `display_standings_on_offday` config option to `config.json.example`
- [ ] Update README
- [ ] Update code as @alphastanley recommended above

ajbowler, 2018-03-05:
@alphastanley we have a Slack channel in the README if you're interested in dropping by

PR #77 (2018-03-12): Add option to display standings on an offday.
If there are no games for the day, display the current standings instead. Turn this off with the "display_standings_on_offday" config option.

Closes #73
commit 11a35fac (2018-03-12): Add option to display standings on an offday. Closes #73 (#77)
closed 2018-03-12 by ajbowler