MLB-LED-Scoreboard/mlb-led-scoreboard #83: Scoreboard crashes when the game of the preferred team is postponed

IsThisThingStillOn, 2018-03-10:
The Red Sox @ Twins game for today is postponed and the scoreboard crashes right away when one of them is configured as the preferred team.

Error message:
`Traceback (most recent call last):
  File "main.py", line 42, in <module>
    GameRenderer(matrix, matrix.CreateFrameCanvas(), games, config).render()
  File "/home/pi/mlb-led-scoreboard/renderers/games.py", line 101, in render
    self.__refresh_game(game, overview)
  File "/home/pi/mlb-led-scoreboard/renderers/games.py", line 185, in __refresh_game
    ScoreboardRenderer(self.canvas, scoreboard).render()
  File "/home/pi/mlb-led-scoreboard/renderers/scoreboard.py", line 15, in render
    InningRenderer(self.canvas, self.scoreboard.inning).render()
  File "/home/pi/mlb-led-scoreboard/renderers/inning.py", line 22, in render
    self.__render_inning_break()
  File "/home/pi/mlb-led-scoreboard/renderers/inning.py", line 40, in __render_inning_break
    num  = self.inning.ordinal()
  File "/home/pi/mlb-led-scoreboard/data/inning.py", line 14, in ordinal
    return Inning.ORDINAL[self.number]
TypeError: list indices must be integers, not str`

ajbowler, 2018-03-10:
Strange, we thought we accounted for this when testing with a past season game but it's great that this was caught early before the season starts.

Much appreciated for finding these bugs today, we'll fix this one for 1.1.0 as well, tentative date to get these in order is 3/29 when the season starts.

IsThisThingStillOn, 2018-03-10:
It also crashes when you chose another preferred team and then it rotates into that game.

ajbowler, 2018-03-10:
Yep that's what I'd expect. It looks like a really simple fix.

ajbowler, 2018-03-13:
Ah, so it's actually marked as `Cancelled`. http://gd2.mlb.com/components/game/mlb/year_2018/month_03/day_10/gid_2018_03_10_bosmlb_minmlb_1/linescore.xml

(It even provides a `reason` attribute which is neat I suppose.)

My guess is this is a spring training quirk because the code we added around postponed games was tested against a regular season game from 2017.

PR #85 (2018-03-13): Handle cancelled games in spring training
Closes #83
commit afa93505 (2018-03-13): Handle cancelled games in spring training (#85)

Closes #83
closed 2018-03-13 by swemoney