MLB-LED-Scoreboard/mlb-led-scoreboard #88: Handle Doubleheaders

ajbowler, 2018-03-13:
If a preferred team has two games on a given day, prioritize displaying the game that is currently in progress (first game was played earlier, second game is currently underway).

Currently we show the first of the two games.

demiller9, 2019-08-07:
My favorite team played a double header today. Neither game was tracked, the scoreboard cycled through the schedule/final all day long.

With some care, you might be able to fix this and handle teams who play successively. (I have a couple of teams in my preferred list. Why can't it follow each game if they are at different times?)

ajbowler, 2019-08-07:
This issue I actually forgot to close, it should be fixed with the latest version (3.3.0) @demiller9 can you verify 3.3.0 works/doesn't work?

demiller9, 2019-08-07:
Yes, I am running 3.3.0 (built / installed about two weeks ago).

closed 2022-04-13 by twfarley