MLB-LED-Scoreboard/mlb-led-scoreboard #89: 128x32 Support

flemingjw, 2018-03-13:
I'm not sure if it would push the pi too hard but would it be possible to display the normal double wide game on 64x32 and the standings page next to it?

ajbowler, 2018-03-13:
Should be totally doable, though I think there is some work that needs to be done, mainly refactoring the shit out of the main game renderer loop.

We should also start working towards storing the coordinates of each rendered component off somewhere so these can be easily mapped depending on the user's board dimensions.

ajbowler, 2018-03-14:
@flemingjw do you have a 128x32? (does that exist? If not, probably just a pair of 64's) I'd love to add support for it but only if there are people that would use it. I'd need somebody to help test since I currently only have a 32x32 in my possession.

flemingjw, 2018-03-14:
I have 4x 32x32 panels

joelguth, 2018-04-27:
I have a pair of  64x32 panels in a chain, which is effectively 4 x 32x32 matrices in a chain. I can create an w128h32.json file with the default settings from w64h32.json.example file and things display just fine, they're just squished. As you mentioned, the ability just to remap the components to a different set of coordinates would allow us to split half of the board up for score information, and the other half for bases, outs, inning, etc. Or something along those lines.

ajbowler, 2018-04-27:
@joelguth at the moment, you have (mostly) full control over the coordinates of things, save some x coords for things that require text-centering. https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/tree/master/ledcoords#custom-coordinates

I currently lack the number of boards to test out 128x32, but if you're willing to give it a shot, I'd definitely welcome any pull requests for it with some example screenshots. I don't expect this would take more than a new `w128h32.json.example` for us to confidently say we support these dimensions, then we can build off of what a standard 128-wide display "should" have present on it from there.

joelguth, 2018-04-27:
Yeah, the default w128h32.json was enough to make it functional, there's just underutilized space that I think could be re-arranged. I've been able to move most things around by putting the team names and scores on the left half, and bases and whatnot on the right. I also increased the base size which displays nicely on the larger space. The challenge is mostly around the team colors that automatically span the length of the board, as well as all of the centered text. I think it would beneficial to be able to relocate the x coordinates of the text such as inning breaks and game status in the json file, as well as set a specific cutoff for how far the team name and score color banners span (right now I have just hard coded it.) I'll keep playing around with it and try to come up with something user friendly.

Fantastic work on this by the way. I built a replica Wrigley Marquee for my dad a year or two ago using a 64x32 matrix and he'll love the ability to see live scores on it now.

ajbowler, 2018-04-27:
Sounds good, let me know if you've got any questions. I'll gladly review any pull requests around this.

closed 2018-05-05 by ajbowler