MLB-LED-Scoreboard/mlb-led-scoreboard #91: Scoreboard is stuck if a game ends early

IsThisThingStillOn, 2018-03-20:
After the game of my preferred team ended, the scoreboard started to rotate through games until it rotated into the  TOR @ PHI game, which was called after 7 innings. It's Right now it's stuck at this game and won't rotate any further.

https://i.imgur.com/IzpNwx2.jpg

My config:
`{
        "preferred_team": "Red Sox",
        "preferred_division": "AL East",
        "display_standings": false,
        "display_standings_on_offday": false,
        "rotate_games": true,
        "rotate_rates": { "live": 15.0, "final": 15.0, "pregame": 15.0 },
        "stay_on_live_preferred_team": true,
        "scroll_until_finished": true,
        "slower_scrolling": false,
        "debug_enabled": false
}
`

ajbowler, 2018-03-20:
New game status: "Completed Early"
Reason: "Rain"

http://gd2.mlb.com/components/game/mlb/year_2018/month_03/day_20/gid_2018_03_20_tormlb_phimlb_1/linescore.xml

Shouldn't be too hard to handle this. If it's completed early, probably just need to show final.

ajbowler, 2018-03-20:
@IsThisThingStillOn issue is fixed in `dev`, I'll just cut release 1.1.1 since it was a quick fix.

PR #92 (2018-03-20): Handle games that were completed early
Fixes #91
commit 5bcee483 (2018-03-20): Handle games that were completed early

Fixes #91
closed 2018-03-20 by ajbowler