MLB-LED-Scoreboard/mlb-led-scoreboard #99: Scoreboard showing TBD for probable starters on all Opening Day games and beyond

IsThisThingStillOn, 2018-03-29:
Installed mlbgame from master branch. It worked fine during Spring Training.

flemingjw, 2018-03-29:
Mine never worked during spring training and I'd assumed it was some exhibition related glitch.
Still don't work

ajbowler, 2018-03-29:
From a discussion in Slack it appears that the returned data in the regular season changed causing a conversion error so it defaulted back to TBD. I don't have time to get to it in the next couple days but @swemoney might.

Really need that mlbgame release, too, though.

ajbowler, 2018-03-31:
Closed with the 1.1.2 release.

closed 2018-03-31 by ajbowler

flemingjw, 2018-04-03:
Perhaps I'm missing something here but I still get this behavior. Every pitcher comes up as "TBD". 
I do recall seeing a pitchers name on a postgame screen on the last day of spring training.

IsThisThingStillOn, 2018-04-03:
You need to install mlbgame from master

flemingjw, 2018-04-04:
I must not be doing it right but I ran pip uninstall on mlbgame, and reinstalled it. Both times it came back with version 2.4.2 or something. It seems to have fixed it for postgame scrolling, but I still get TBD vs TBD returned for pregame.

flemingjw, 2018-04-05:
That did the trick @ajbowler 
Thanks guys

PR #100 (2018-03-30): Fix issue when a pitcher hasn't recorded an out yet
closes #99