MLB-LED-Scoreboard/mlb-led-scoreboard #117: Refactor the Game Renderer loop

ajbowler, 2018-04-10:
Tagging @swemoney 

This is where we can discuss the grand scheme to refactor the big loop in the game renderer. We could probably start by abstracting away some of the fields like the current scrolling text position and the flag for the data needing a refresh, but will probably take some research.

swemoney, 2018-04-11:
I'm thinking one of the first things we can refactor that might not be too difficult would be separating the mlbgame calls out of the renderer and into it's own object. Ideally, we create this object at the beginning when the scoreboard starts and the data object has it's own logic to refresh the data on a 10-15 second interval.

After this is done, the game loop can be reduced to always being a 0.1 second refresh and it'll just pull data from the data object that should be refreshing itself.

These 2 things are my big overarching idea for how the renderer and data object should interact when all is done. I haven't gotten enough of a chance to go through thinking of the best way to actually implement it all, though.

ajbowler, 2018-04-13:
An observer pattern maybe? I'm not very good with design patterns let alone in Python.

swemoney, 2018-04-27:
I'll add to this conversation for the refactor. The initial overall refactor is mostly done but there are still pieces that can use some work. Not necessary for a release but things I would like to get done soon.

- [ ] Standings and error renderers should be updated to the same class format as the rest of the renderers.
- [ ] Starting the standings renderer should  probably be moved into the main renderer where it makes more sense. This can also give us the ability to add an option to include the standings into the rotation of games in the future.
- [ ] All renderers can be cleaned up a bit to only require the `canvas`, `data` and `scrolling_text_pos` to be passed.
- [ ] A lot of other methods can be converted to accept the `data` object instead of separate `overview` or `game` objects.
- [ ] GameRenderer class has been replaced with the MainRenderer. File can be removed when comfortable everything is working great.

swemoney, 2018-04-28:
I agree on both counts. The old game renderer should be removed before we push 1.3.0 for real. And the config file needs a lot of work. I wouldn't want to maintain backwards compatibility or anything so I would be fine throwing that in 2.0.0

ajbowler, 2018-04-29:
@swemoney I'm gonna close this issue since the main goal here has been accomplished (reworking the game renderer) and it's looking great. Your checklist above is still valid and could probably have separate issues, but I'd say this particular one is good to go.

closed 2018-04-29 by ajbowler