MLB-LED-Scoreboard/mlb-led-scoreboard #128: Support for two parallel 16x32 panels

phonorad, 2018-04-27:
I am trying to get the program running using two 16x32 displays.   They are 1:4 multiplexed displays and I am using the zheller hat which supports parallel panels (up to 3).  I was able to get 2 panels (32x32) working with the zheller demos (clock, Demo D=0, etc).  I need to use led-multiplexing=4, which does not seem to currently be supported by the MLB program (only takes up to =3 without error), so I modified the MultiplexMapper.cc to have that multiplexing type be =3 instead.  The demos still work, but when I use "sudo python main.py --led-rows=16 --led-parallel=2 --led-multiplexing=3", the board does not look at all correct.  Should this configuration be working or are some other changes to the MLB scoreboard program required?

ajbowler, 2018-04-27:
@phonorad this looks to be a recent enhancement to the rgbmatrix library: https://github.com/hzeller/rpi-rgb-led-matrix/commit/d25e9b6a2d0fa1879927ed18780b27e8464352f7

I'll need to get our project's submodule updated, it currently dates back to a commit on January 27th: https://github.com/hzeller/rpi-rgb-led-matrix/commit/e5b41259279095ce79ac8a8b3332aad35d9f5d79

phonorad, 2018-04-27:
Great, thanks, here are a few screenshots of the scrambled scoreboard, along with the working demo and appropriate command lines used, just in case they are of any use.

[image]
[image]
[image]
[image]

phonorad, 2018-04-27:
My ultimate goal, by the way, is to make led-chain=2, using 4 total boards for a 32x64 total size, but I wanted to start with the 32x32 configuration and get it working first.  Thanks again!

ajbowler, 2018-04-28:
@phonorad I've pushed this change to the `update-rgbmatrix` branch. Could you check this branch out and verify you can set the Z-strip flag?

phonorad, 2018-04-28:
I cloned the branch, and then re-installed.  The Matrix/examples-api-use work, but when I run the program: "pi@raspberrypi:~/mlb-led-scoreboard $ sudo python main.py"
I get:
"INFO (18:31:59): MLB LED Scoreboard - v1.3.0"
and then it goes back to the command prompt, and no LED action.

When I run the original version (in renamed directory mlb-led-scoreboard_old), the program still runs and lights up the panel, although with the wrong mapping, of course.

ajbowler, 2018-04-28:
What kind of connection are you using? If you're running off of Adafruit, you'll need to supply an led-gpio-mapping flag to the command as well. I've gotten a blank screen when running without that flag by mistake

phonorad, 2018-04-28:
Sorry, I forgot to make a new config.json file.  Got is working...almost.  Get most of the screen right, but still a few things out of place.  Will post a pic shortly.  Thanks for your efforts thus far!

phonorad, 2018-04-28:
OK, it takes the led-multiplexing option of 4. 
Now, with the command line of: 
sudo ./demo -D 3 --led-rows=16 --led-parallel=2 --led-multiplexing=4 --led-brightness=25
I get a correct (I think) pattern of;
[image]
When I run the scoreboard program with a command line of;
sudo python main.py --led-rows=16 --led-parallel=2 --led-multiplexing=4 --led-brightness=50
I get;
[image]
which is partially correct (this is the Tampa Bay Boston game going on now).  Any thoughts?

phonorad, 2018-04-28:
I should note that I did have to modify multiplex-mappers.cc in the lib directory and then did a 'make' from the Matrix directory.   This make the demos work properly with my panels.  Do I need to do anything else to get the scoreboard program to recognize the change?

The following was changed in multiplex-mappers.cc:
Original Code:
   result->push_back(new ZStripeMultiplexMapper("ZStripe", 0, 8));
Changed to:
  result->push_back(new ZStripeMultiplexMapper("ZStripe", 8, 0));

ajbowler, 2018-04-28:
At this point you might reach out to the rgbmatrix library maintainers. Since you've got a different wiring that I'm unfamiliar with, he should know exactly what to do here. I can only really go so far as helping you install the scoreboard itself

phonorad, 2018-04-29:
OK, thanks. Just so I understand, if I make a change to a matrix/lib program (one of the .cc files) and then recompile then with make, do I need to do anything to the MLB Scoreboard program like rebuild it or anything, to make it recognize the change?

ajbowler, 2018-04-29:
You might need to redo the step `sudo pip install -e python/`

phonorad, 2018-04-29:
Thanks, I will try that.

ajbowler, 2018-04-29:
I'm gonna close this issue since I updated our LED flags to match the matrix library's current implementation, drop by the Slack channel if you continue to run into issues. I'm hoping the rgbmatrix maintainers can help you out with your wiring setup, that seems right up their alley.

PR #139 (2018-04-29): Update rgbmatrix submodule
Closes #128
commit f8939710 (2018-04-29): Update rgbmatrix submodule (#139)

Closes #128
closed 2018-04-29 by ajbowler