MLB-LED-Scoreboard/mlb-led-scoreboard #138: RPC Server for remote control

swemoney, 2018-04-28:
I've got a proof of concept working for this and am really excited about the possibilities it'll bring. Just waiting on some of the recent refactoring to get done before going ahead full steam with it. The RPC server will allow controlling most aspects of the scoreboard remotely via any RPC client. This opens up the possibility for a simple web based front end, alexa support (ala #61), and much more that I the community can do to enhance and automate their scoreboard completely. I'd love to start a discussion about what kind of ideas people have for usage so I know exactly what methods we need to expose.

ajbowler, 2018-04-28:
This is awesome! cc @craigm26 

As far as what I'd expect to be able to do with Alexa, I'd like to be able to say "give me the <team> game" and it'll set the scoreboard to that game. I'd also like to do something similar for the standings.

angelogoncalve, 2018-05-10:
Use this https://github.com/joshmarshall/jsonrpctcp

closed 2022-04-08 by twfarley