MLB-LED-Scoreboard/mlb-led-scoreboard #176: Team records not properly showing in 5x7 font in Standings

WillWillD, 2018-06-05:
[image]

**Below is the Coord layout file**

{
  "defaults": {
    "font_name": "5x7"
  },
  "bases": {
    "1B": {
      "x": 54,
      "y": 22,
      "size": 8
    },
    "2B": {
      "x": 49,
      "y": 17,
      "size": 8
    },
    "3B": {
      "x": 44,
      "y": 22,
      "size": 8
    }
  },
  "final": {
    "inning": {
      "x": 32,
      "y": 22
    },
    "scrolling_text": {
      "x": 0,
      "y": 31,
      "width":64
    },
    "nohit_text": {
      "x": 1,
      "y": 25
    }
  },
  "inning": {
    "break": {
      "number": {
        "x": 32,
        "y": 29
      },
      "text": {
        "x": 32,
        "y": 22
      }
    },
    "number": {     
      "x": 50,
      "y": 21
    },
    "arrow": {
      "size": 4,
      "up": {
        "x_offset": -5,
        "y_offset": -4
      },
      "down": {
        "x_offset": -5,
        "y_offset": -2
      }
    }
  },
  "outs": {
    "1": {
      "x": 1,
      "y": 26,
      "size": 5,
      "nohit": {
        "x": 1,
        "y": 26,
        "size": 5
      },
      "perfect_game": {
        "x": 1,
        "y": 26,
        "size": 5
      }
    },
    "2": {
      "x": 8,
      "y": 26,
      "size": 5,
      "nohit": {
        "x": 8,
        "y": 26,
        "size": 5
      },
      "perfect_game": {
        "x": 8,
        "y": 26,
        "size": 5
      }
    },
    "3": {
      "x": 15,
      "y": 26,
      "size": 5,
      "nohit": {
        "x": 15,
        "y": 26,
        "size": 5
      },
      "perfect_game": {
        "x": 15,
        "y": 26,
        "size": 5
      }
    }
  },
  "batter_count": {
    "x": 4,
    "y": 23,
    "nohit": {
      "x": 4,
      "y": 23
    },
    "perfect_game": {
      "x": 15,
      "y": 21
    }
  },
  "nohitter": {
    "x": 25,
    "y": 30,
    "innings_until_display": 6
  },
  "pregame": {
    "matchup": {
      "x": 32,
      "y": 6
    },
    "scrolling_text": {
      "x": 0,
      "y": 30,
      "width": 64,
      "warmup": {
        "x": 0,
        "y": 30,
        "width": 64
      }
    },
    "start_time": {
      "x": 32,
      "y": 20
    },
    "warmup_text": {
      "x": 32,
      "y": 20
    }
  },
  "standings": {
    "offset": 6,
    "height": 32,
    "width": 64,
    "divider": {
      "x": 0
    },
    "stat_title": {
      "x": 40
    },
    "team": {
      "name": {
        "x": 1
      },
      "stat": {
        "x": 64
      }
    }
  },
  "status_text": {
    "x": 32,
    "y": 20,
    "short_text": false
  },
  "teams": {
    "background": {
      "away": {
        "width": 64,
        "height": 7,
        "x": 0,
        "y": 0
      },
      "home": {
        "width": 64,
        "height": 7,
        "x": 0,
        "y": 7
      }
    },
    "name": {
      "away": {
        "x": 1,
        "y": 6
      },
      "home": {
        "x": 1,
        "y": 13
      }
    },
    "runs": {
      "away": {
        "x": 62,
        "y": 6
      },
      "home": {
        "x": 62,
        "y": 13
      }
    }
  }
}

ajbowler, 2018-06-05:
Thanks @WillWillD I'll try and have a look soon. It's looking like the name needs to move a bit more and the records are too far to the left. Probably need to account for the different font size.

ajbowler, 2018-06-05:
Related: https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/pull/177

swemoney, 2018-06-06:
#177 will add a little bit more customization to the standings file. The biggest issue is that the offset can't be greater than 6 or we can't fit the entire division on the screen. With this PR, you have more control over the exact location of the `record` and `games_back`.

My best idea to resolve this issue if you want a 5x7 font is to change the divider color to the same thing as the background color (#177 also modifies the way the dividers render, and draws them first so you can draw text on top of them) and then align the text as you'd like without any dividers. This saves a little bit of space and you should still be able to fit the 5x7 font with a pixel or so between each line.

ajbowler, 2018-07-11:
@WillWillD please check out the dev branch. https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/pull/184 shows the changes.

Basically, switch out `stat` with `record` and `games_back`. This gives you a bit more control over how the standings look.

closed 2018-07-22 by ajbowler