MLB-LED-Scoreboard/mlb-led-scoreboard #180: LED matrix only showing one color

ckomlos, 2018-06-21:
When I go to run the actual scoreboard program the LED matrix only lights up one color. I have run the python sample programs and those show all three colors but the real program only lights up the red channel. 

Like this:[image]

This is the graphics.py sample: 
[image]

The code I am running is: `sudo python main.py --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=2`

This is the config file I have: 
[config.json.txt](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/2121952/config.json.txt)

ckomlos, 2018-06-21:
This is the matrix I am using:  https://www.aliexpress.com/item/High-Refresh-Rate-Indoor-P6-RGB-SMD3528-Full-Color-32x32-Pixels-LED-Module-192x192mm-for-P6/32763763208.html?spm=a2g0s.9042311.0.0.40694c4dPUdfPH

ckomlos, 2018-06-21:
Underside of the board

[image]
[image]

ajbowler, 2018-06-22:
I will reach out to the rgbmatrix maintainer and see if there's anything at the library level we can account for here.

ajbowler, 2018-06-27:
Most of the discussions have been in Slack, the `--led-pwm-bits` command appears to be having some effect here. Will need to keep playing around and see what options this specific board needs to display correctly.

closed 2018-06-27 by ajbowler