MLB-LED-Scoreboard/mlb-led-scoreboard #183: 32x32 board flickering

trevordavies095, 2018-07-06:
So my board is all setup and I absolutely love it! However, the RGB board sort of flickers like a bad refresh rate. Anyone ever have this issue?

ajbowler, 2018-07-06:
There are a few ways to help this out. Are you using an Adafruit HAT?

Try adding `--led-gpio-slowdown=2` to your command when running the board. That's a common solution people use. It's not perfect, but it's typically much better.

trevordavies095, 2018-07-06:
That gave me an unrecognized argument.

ajbowler, 2018-07-06:
Drop by the Slack channel and you might get some more ideas. The slowdown flag is just the one I'm familiar with.

Slack link is in the README. If we find a good solution in there I'll document it in this issue.

closed 2018-07-08 by trevordavies095

ajbowler, 2018-07-08:
@trevordavies095 did you find a working resolution?

trevordavies095, 2018-07-09:
I used `--led-pwm-bits=8` and it reduced it slightly. It's not that big of a problem as I'm not constantly staring at the scoreboard,