MLB-LED-Scoreboard/mlb-led-scoreboard #191: pregame text not scrolling

bojanggles14, 2018-07-20:
I can't seem to get any of the scrolling text to work.  everything else is fine, I had to adjust the placement of the start time, but other then that the only thing I have changed is the preferred team settings.  Here is my JSON file.

ajbowler, 2018-07-20:
I don't see a JSON file :)

bojanggles14, 2018-07-20:
My bad i didn't see that .json wasn't supported in the drag and drop
[w128h32.json.txt](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/2215251/w128h32.json.txt)

ajbowler, 2018-07-20:
ah, 128x32. This is something you'll have to be patient with us with, as our support for 128x32 is really limited. I don't personally own the hardware to test that out myself.

@joelguth are you able to take a look at this?

joelguth, 2018-07-20:
Confirmed.

Change the x coords to 0 for the pregame portion:

  "pregame": {
    "scrolling_text": {
      "x": 0,
      "y": 31,
      "width": 128,
      "warmup": {
        "x": 0,
        "y": 31,
        "width": 128
      }

bojanggles14, 2018-07-20:
that fixed it, thanks! The only comment i have is that every game shows TBD VS TBD for the pitchers, even the ones starting in less then 30 min, is this common?

ajbowler, 2018-07-20:
Yes, I pushed a fix for that to mlbgame a while ago, but they have not yet released it.

If you're feeling adventurous you can reinstall the mlbgame module from their master branch, otherwise you'll have to wait till they release a new version.

closed 2018-07-20 by ajbowler