MLB-LED-Scoreboard/mlb-led-scoreboard #197: 128x64 support

jasonbogovich, 2018-08-13:
Hello, thanks for doing this. How much work is adding this support? I've just started playing with the RPI matrix python lib in order to make a more simple scoreboard for his baseball games but I think it's cool to be a smart display as well.

ajbowler, 2018-08-13:
It would require creating a new layout file for those dimensions and spending some time to make sure nothing looks wonky.

I personally only own a 64x32 and a 32x32 so 128x64 support/maintainability would need to come from another helpful soul. I'll happily review any PRs with provided screenshots.

ajbowler, 2018-08-13:
https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/tree/master/ledcoords for reference

jasonbogovich, 2018-08-13:
Thanks. I am going to try to hack away at it. How do I go about disabling the 128x64 support message?

ajbowler, 2018-08-13:
https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/blob/master/data/scoreboard_config.py#L155

This is printed out when it can't find a file based on the dimensions you provided for your board, in this case, w128h64, which we don't currently have a `.json.example` file for.

If you add a `w128h64.json.example` to `ledcoords/`, which would become the default 128x64 layout, that message will go away because it found something to reference.

jasonbogovich, 2018-08-13:
Thank you AJ, I'll see what I can do.

jasonbogovich, 2018-08-13:
With the exception of the bottom 32 pixels running with only bgcolor the 128x32 config works just fine.  I am waiting till games start so I become more acquainted with features as is. Thanks to everyone for this work, I love this thing, I've spent a lot of time trying to encapsulate OOP with this Matrix already so I totally appreciate the work that you all did here. What would you all change about a 128x64 layout, I'm sure you've spent more time thinking about it.

ajbowler, 2018-08-13:
@joelguth @swemoney any thoughts? I don't really think there's much to be done here honestly, maybe stretch the team banners more to fill the extra space?

jasonbogovich, 2018-08-14:
Follow two games at once, or use the bottom 128x32 to run a concurrent and perhaps alternating second choice/s game? I love the minimalism and spatially mindful functionality. I think the layout is great and maybe I just need to learn some of the options better first.

jasonbogovich, 2018-08-14:
[image]
I'll get a better shot but it's a thing of beauty. I wish I had a nice 3d printer for it.

jasonbogovich, 2018-08-14:
Is pitcher name and/or batter name provided?

ajbowler, 2018-08-14:
Pitcher information I'm not sure, will need to dig into it. Displaying two games at a time I've thought of before but that would be a huge undertaking as it imposes design choices for configuration, layouts, and making sure all pieces work like scrolling text.

Both of those are/should be eparate issues, at the very least having a default layout for 128x64 would be sufficient to close this issue.

jasonbogovich, 2018-08-14:
What about MLB's full scoreboard? That was my other thought for a way to improve this layout.

jasonbogovich, 2018-08-14:
Or I can just try to sketch out a workable solution as is with resizing for starters? Full scoreboard by inning would be sweet though.

ajbowler, 2018-08-14:
Anything dealing with new data or functionality should be done as a separate issue(s) that need to go through their own discussion and scrutiny before we can consider adding them in.

For this just a new layout file for these dimensions is enough, one thing at a time.

ajbowler, 2018-08-14:
We can iterate on the layout as we add new features, but just having one to begin with is a huge step.

ajbowler, 2018-08-14:
Displaying two games at a time is #62 FWIW. Let's get this new layout rolling then we can do the more involved things to follow.

jasonbogovich, 2018-08-14:
[image]
So as far as I can see it's a very minor update for the 128x32 config. Just copy the 128x32 file and modify the following lines with "width":64, -> 128. Unless you think resizing elements would be useful in the long run?

```
  "teams": {
    "background": {
      "away": {
        "width": 128,
        "height": 11,
        "x": 0,
        "y": 0
      },
      "home": {
        "width": 128,
        "height": 11,
        "x": 0,
        "y": 11
      }
    },
```

jasonbogovich, 2018-08-14:
[image]
Here is what my other scoreboard looks like. I thought it would be awesome to have a scorekeeper app that also does some player and team stats and drives a scoreboard with single user input. Gamechanger charges a ransom for stats each season.

jasonbogovich, 2018-08-14:
That was just a first draft though, I like the ESPN style for bases that you utilize.

ajbowler, 2018-08-14:
That scoreboard looks awesome! Very old school.

I think I like how you have it for pregame, maybe move the pitching matchup text down 4-5 pixels.

Will need to see how an in-game one looks with the bases and pitches.

jasonbogovich, 2018-08-14:
Does the team banner reformat to 64x11 for games or will the stripes end up on the gametime stats? I will move the pitching down. Is scroll speed 5 possible?

jasonbogovich, 2018-08-14:
Gotta get some stuff done, I'll be back in a bit to work on this more.

joelguth, 2018-08-14:
I would probably format it more like a 64x32 panel and use a larger font since you have all that space.

joelguth, 2018-08-14:
Yeah, those get pulled down from the rpi LED matrix library. I typically have to edit them with a BDF font editor to get them to my liking.

jasonbogovich, 2018-08-14:
[image] Go go gadget game simulator.

jasonbogovich, 2018-08-14:
Maybe bigger font? Which element produces the TBD? I might have misplaced it XD

ajbowler, 2018-08-14:
I think I like that font.

The scrolling text and such should all be under `pregame` in the layout

jasonbogovich, 2018-08-14:
[image] I'm getting there. Still a lot of tweaking to go.

jasonbogovich, 2018-08-15:
I should be done with this soon, I've not tested perfect game stuff and maybe need to test it a night or two, I might also try to run a config for 64x64 so that it's supported.

ajbowler, 2018-08-15:
We can open 64x64 as a separate issue if needed

ajbowler, 2018-08-15:
I personally like the bold ones more but I'm fine with either.

jasonbogovich, 2018-08-16:
Sorry for another off topic. Maybe I should find a general chat, but have you thought about or attempted implementing https://teamcolorcodes.com/mlb-color-codes/ and resorted back to the pastel color scheme?

ajbowler, 2018-08-16:
The README has a Slack link at the top, you can join it from there, there's a #dev channel

PR #200 (2018-08-31): 128x64
Closes #197 

This layout file was taken from @jasonbogovich, just committing it for him.

I've tested it on my 64x32 to make sure nothing broke but I can't really verify this on my own
closed 2018-09-01 by ajbowler