MLB-LED-Scoreboard/mlb-led-scoreboard #199: 64x64 support

ajbowler, 2018-08-20:
Add a default layout for 64x64

MrDennisM, 2019-02-25:
Has anyone had an opportunity to work on a 64x64 layout?

I'm new to GitHub so I apologize in advance if this is posted wrong here.  Trying to build this scoreboard for my son.  Can't even get the Pi OS loaded but I'll get there!  Thank you.

swemoney, 2019-02-25:
We haven't officially supported 64x64 yet but it's pretty trivial to add. If you'd like to help, you can create a copy of the 32x32 design and modify it to fit the larger matrix. In the `ledcoords` directory, you can copy the `w32h32.json.example` to a new file called `w64h64.json.example`. This will allow the scoreboard to start up. Then it's just a matter of modifying the coordinates in that file until everything looks great.

ajbowler, 2019-02-25:
The biggest hurdle for 64x64 is just having someone that owns a board of that dimension create the layout for us. None of the official project maintainers owns one of them, so we'll have to rely on the community to work with us on that.

pawptart, 2019-10-15:
Not to plug my own stuff, but I have a 64x64 board set up as an MLB scoreboard as well and would be willing to look into this at some point. Mine is not nearly as fleshed out as yours.

https://github.com/pawptart/MiniTron

kwhite1956, 2020-01-10:
I have a 64x64 panel coming and will work on this once it arrives

ajbowler, 2020-03-05:
Closed by #282

PR #282 (2020-03-05): Add a default layout for 64x64
Closes #199. Adds a default layout for 64x64 LED matrices.
closed 2020-03-05 by ajbowler