MLB-LED-Scoreboard/mlb-led-scoreboard #203: Preferred Division Not Displaying

DanGarion, 2018-11-09:
Is anyone else having issues with the preferred division not displaying? Here is my config. But I'm not getting the NL West, I'm only seeing NL Central... Getting tired of seeing those standings during the offseason and planning to shut things down for a few months, but wanted to figure out what was wrong first.

> {
>         "preferred_team": "Dodgers",
>         "preferred_division": "NL West",
>         "display_standings": false,
>         "display_standings_on_offday": true,
>         "rotate_games": true,
>         "rotate_rates": { "live": 15.0, "final": 15.0, "pregame": 5.0 },
>         "stay_on_live_preferred_team": true,
>         "scroll_until_finished": true,
>         "end_of_day": "00:00",
>         "display_full_team_names": true,
>         "slower_scrolling": true,
>         "debug_enabled": false
> }
> 
> 
>

swemoney, 2018-11-09:
preferred_team and preferred_division both take an array. So try `["NL West"]` instead. You probably should wrap your preferred team in brackets as well.

swemoney, 2018-11-09:
And I just thought of it, if you're on the most recent version, that config file isn't going to work anymore. Copy the .example into your old config (overwrite it). The config file changed with version 2.0.

DanGarion, 2018-11-12:
Hah... that would probably be my issue. :)

Thanks!

closed 2018-11-12 by DanGarion