MLB-LED-Scoreboard/mlb-led-scoreboard #211: 128x64 not working with default config.json

phonorad, 2019-01-30:
I am attempting to upgrade from by 64x32 P8 (four 32x16 panels) to 128x64 P4 (four 64x32 panels) to get better display resolution.   When I run one panel (63x32) all is fine, but when I run "sudo python main.py --led-cols=64 --led-chain=2 --led-parallel=2" using the default config.json file, I get the following error;
Traceback (most recent call last):
  File "main.py", line 36, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 42, in render
    self.__render_standings()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 81, in __render_standings
    self.__render_offday()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 67, in __render_offday
    scroll_max_x = self.__max_scroll_x(self.data.config.layout.coords("offday.scrolling_text"))
  File "/home/pi/mlb-led-scoreboard/data/layout.py", line 42, in coords
    raise e
KeyError: 'offday'

Perhaps something in the scrolling text for the news seems to not like the display dimensions?
Thanks for any help you can provide!

ajbowler, 2019-01-30:
Looks like that may have gotten missed in the 128x64 config as that whole `offday` object is missing.

I'll investigate, problem with 128x64 is I don't have a board with that dimension so whatever configs I add to that file may/may not look quite right so some tweaking to the numbers might be required.

If you're willing, try adding this below the `teams` object in `ledcoords/w128h64.json.example`:

```
  "offday": {
    "scrolling_text": {
      "x": 0,
      "y": 30,
      "width": 64
    },
    "time": {
      "x": 40,
      "y": 9
    },
    "conditions": {
      "x": 40,
      "y": 17
    },
    "temperature": {
      "x": 10,
      "y": 23
    },
    "wind_speed": {
      "x": 0,
      "y": -2
    },
    "wind_dir": {
      "x": 0,
      "y": -2
    },
    "wind": {
      "x": 40,
      "y": 23
    },
    "weather_icon": {
      "x": 2,
      "y": 1,
      "width": 15,
      "height": 15
    }
  }
```

swemoney, 2019-01-30:
Ya. I'm not sure how I missed this. You can also create a new file in `ledcoords` called `128x64.json` with just the extra stuff.
```
  {
  "offday": {
    "scrolling_text": {
      "x": 0,
      "y": 30,
      "width": 64
    },
    "time": {
      "x": 40,
      "y": 9
    },
    "conditions": {
      "x": 40,
      "y": 17
    },
    "temperature": {
      "x": 10,
      "y": 23
    },
    "wind_speed": {
      "x": 0,
      "y": -2
    },
    "wind_dir": {
      "x": 0,
      "y": -2
    },
    "wind": {
      "x": 40,
      "y": 23
    },
    "weather_icon": {
      "x": 2,
      "y": 1,
      "width": 15,
      "height": 15
    }
  }
```

phonorad, 2019-01-30:
Thanks for the fast reply! I will give it a try tomorrow evening!

phonorad, 2019-01-31:
I edited the ledcoords/w128h64.json.example file to add the code and it is working, but as you indicated, some of the object positions will need to be adjusted.  I pretty much double the value of most x,y coordinates, except the weather icon.  The wind speed and wind positions look good, but I don't understand their coordinate system versus the other objects.  At any rate, I will play with the settings and post when I get a layout that looks good, unless swemoney beats me to it!

[image]

phonorad, 2019-01-31:
Sorry, didn't mean to close issue!

ajbowler, 2019-01-31:
Much appreciated, thanks for testing this out! Please post your layout in here when you're done and we can port those over as the new default. We don't currently have an in-house support for 128x64 so this is incredibly valuable.

phonorad, 2019-01-31:
Will do!  Unfortunately extremely busy at work right now, but will work on it and post a config as soon as I can.  I will certainly be doing the same as play starts, so will post any of those updates as well.  Thanks!

ajbowler, 2019-02-20:
@phonorad have you had any time to test this out?

phonorad, 2019-02-21:
Yes, I have a new layout for the 128x64 board, but could not figure out how to do a pull request.  I had checked out the 'bugs' branch (I think), made mods to the layout for the 128x64 in the ledcoords folder, but then got stuck and am not sure what to do from there (not sure how to do a pull request and what to do with the files, etc).  I am not sure if the scoreboard will show live games for the early spring training games (first Red Sox game is tomorrow), but I will check that as soon as I see a game and can also change the layout for that if needed.
On another note, I keep getting crashes on the weather app when the program goes to get the latest weather update.  Too many retries and the program quits. I am not sure if network congestion at work is causing it or not, but it would be good if there was a way to have the program not stop when this happens, just keep going and try again next update.  Not sure if this is in the mlb-led-scoreboard app or some other module, so this may be easier said than done.
I am also getting a 'could not retrieve standings' message when the program runs.  I assume this is because there are no standings to display at this point, but am mentioning it in case there is something else going on because I remember the standing being available for quite a while after the season ended.

ajbowler, 2019-02-21:
Please open a new issue for the stuff going on with weather.

The standings message is expected, it's just debug logging. The standings are retrieved for the given day, same as the live games, so there's no standings today cause the season hasn't started.

If you're having trouble opening a PR with your layout changes, just paste your 128x64 layout in here and we can make the PR.

Also please post a picture or two of your board with the 128x64 layout so we can document it in the README

swemoney, 2019-02-21:
ya, the weather stuff is new so if you can make a new issue for it, I'll take a look. I, personally, haven't seen any issues with the weather fetching but I know there are plenty of fringe cases that I haven't thought of yet.

phonorad, 2019-02-21:
I have attached the following to this Email; 1. w128h64.json - This file has my modified layout as shown in the attached picture of the Offday screen 2. scoreboard.json - This contains the color scheme shown in the picture, in case anyone wants to modify their colors the same way 3. img6919.jpg - Picture of the new 128x64 layout 4. Icons 3x30.zip - The 15x15 weather icons were too small on the larger board, so I enlarged them to 30x30 pixels (the 30x30 size weather icons are shown in the picture). Needed to replace the icons in the assets/weather folder with these, perhaps a future version can use the appropriate pixel size icons based on board size selected.
Let me know if the attachments come through OK and if you need anything else, and thanks again for your efforts!

 
 

   
  "defaults": {
    "font_name": "9x18B"
  },
  "bases": {
    "1B": {
      "x": 75,
      "y": 43,
      "size": 18
    },
    "2B": {
      "x": 65,
      "y": 33,
      "size": 18
    },
    "3B": {
      "x": 55,
      "y": 43,
      "size": 18
    }
  },
  "final": {
    "inning": {
      "x": 67,
      "y": 61
    },
      "y": 44,
      "width": 128
    },
    "nohit_text": {
      "x": 1,
      "y": 33
    }
  },
  "inning": {
    "break": {
      "number": {
        "x": 87,
        "y": 44
      },
      "text": {
        "x": 46,
        "y": 44
      }
    },
    "number": {
      "x": 120,
      "y": 45
    },
    "arrow": {
      "size": 8,
      "up": {
        "x_offset": -12,
        "y_offset": -10
      },
      "down": {
        "x_offset": -12,
        "y_offset": -2
      }
    }
  },
  "outs": {
    "1": {
      "x": 2,
[... 362 lines cut ...]
    },
    "wind": {
      "r": 255,
      "g": 128,
      "b": 0
    },
    "weather_icon": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  },
  "default": {
    "background": {
      "r": 7,
      "g": 14,
      "b": 25
    },
    "text": {
      "r": 255,
      "g": 235,
      "b": 59
    }
  }
}

phonorad, 2019-02-21:
Thanks, the next time I see the error I will open a new issue and submit the error message to help determine the cause.

phonorad, 2019-02-21:
I don't think the scoreboard picture and the weather icon zip file came through on my post above, so here they are;
[image]
[Icons 30x30.zip](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/files/2889632/Icons.30x30.zip)

ajbowler, 2019-02-23:
I'm going to file a separate issue re: the bigger weather icons. That's a new feature that may be worth supporting first-class

ajbowler, 2019-02-23:
@phonorad for your custom color, I invite you to add that to our community colors/layout page in our wiki space. Very cool!

phonorad, 2019-02-23:
Excellent, thanks for putting in an issue to perhaps add the larger icons for the higher resolution boards!  I will add my color scheme to the layout page so that others can enjoy.

phonorad, 2019-02-23:
> If you wouldn't mind contributing to making sure the layout looks good once we get live games in a few weeks, that would also be great. Thanks!

Just to close the loop on this, first game today and the existing scoreboard layout for the 128x64 looks great!  Here is a look on my 128x64 board, I think it is good to go.

[image]

closed 2019-02-24 by ajbowler