MLB-LED-Scoreboard/mlb-led-scoreboard #213: Program quits upon failure to get Weather data

phonorad, 2019-02-21:
I believe that the program is quitting is the program fails to retrieve weather data  from weathermap.org.  Not sure what the update time is, but after a few hours on my work network I get the following error.  Using a Pi Zero w with Stretch Lite and WiFi Power Management is off. At home I do not get this error (Pi 3B with Stretch Desktop.  Not sure if it is network congestion or some other issue, but would be good to have the program keep running and try again next update, or something like that.

 
[image]

ajbowler, 2019-02-21:
Looks like a classic timeout from a third party, agreed we probably shouldn't crash, just show the weather data we got from the last successful call

swemoney, 2019-02-23:
I just uploaded a new branch I'd love for you to test out. Also, anyone else having a similar issue, check and see if the issue is still happening. You'll see a warning message in your logs instead of a straight up crash if it's working as intended.

Checkout the `weather-timeout` branch and run your scoreboard like normal and let me know if you notice any connection errors regarding the weather.

If you're not sure how to checkout this branch, you should be able to just do this from your `mlb-led-scoreboard` directory.
```
git fetch
git checkout weather-timeout
```

phonorad, 2019-02-23:
I have checked out the branch on my home board, which does not seem to have the issue, at least not as often, but will leave it running this weekend and see what happens.  Will do the same on my work board, which exhibits the issue regularly, on Monday and report back.  Thanks a lot!

ajbowler, 2019-02-24:
Fixed by https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/pull/223

closed 2019-02-24 by ajbowler