MLB-LED-Scoreboard/mlb-led-scoreboard #215: Error in lxml parser?

phonorad, 2019-02-23:
Both of my scoreboards (64x32 and 128x64) started giving this error today when the program is started.  Perhaps something to do with the start of spring training games?  Anyone else seeing this issue?

Traceback (most recent call last):
  File "/home/pi/mlb-led-scoreboard/main.py", line 36, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 57, in render
    self.__render_game()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 88, in __render_game
    self.data.refresh_overview()
  File "/home/pi/mlb-led-scoreboard/data/data.py", line 113, in refresh_overview
    self.overview = mlbgame.overview(self.current_game().game_id)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/__init__.py", line 198, in overview
    return mlbgame.game.Overview(mlbgame.game.overview(game_id))
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/game.py", line 341, in overview
    output = add_raw_box_score_attributes(output, game_id)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/game.py", line 370, in add_raw_box_score_attributes
    raw_box_score_root = etree.parse(raw_box_score).getroot()
  File "src/lxml/etree.pyx", line 3426, in lxml.etree.parse
  File "src/lxml/parser.pxi", line 1860, in lxml.etree._parseDocument
  File "src/lxml/parser.pxi", line 1880, in lxml.etree._parseFilelikeDocument
  File "src/lxml/parser.pxi", line 1775, in lxml.etree._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 1186, in lxml.etree._BaseParser._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 600, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 710, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 639, in lxml.etree._raiseParseError
  File "http://gd2.mlb.com/components/game/mlb/year_2019/month_02/day_22/gid_2019_02_22_neubbc_bosmlb_1/rawboxscore.xml", line 1
lxml.etree.XMLSyntaxError: Space required after the Public Identifier, line 1, column 55

swemoney, 2019-02-23:
This could have been an issue with MLB just not having any data for this game (Spring Training games are kind of a mixed bag when it comes to reliable data from MLB) or something has changed and will require some work on the `mlbgame` library we use to get our data. We will definitely keep an eye on this as Spring Training continues.

ajbowler, 2019-02-23:
I have a short-term solution. I sent a patch to mlbgame: https://github.com/panzarino/mlbgame/pull/119

I have that fix locally on another branch so I'm gonna let this bake for a bit before I release.

phonorad, 2019-02-23:
Thanks! Just loaded it by running;
  sudo pip uninstall -y mlbgame
  sudo pip install git+git://github.com/ajbowler/mlbgame.git@#egg=mlbgame

closed 2019-02-23 by ajbowler