MLB-LED-Scoreboard/mlb-led-scoreboard #220: Rotation hangs at unknown Final State

twfarley, 2019-02-24:
Rotation hangs with this in debugger:

DEBUG (16:49:22): Overview Refreshed: 2019/02/23/clemlb-cinmlb-1
DEBUG (16:49:22): Pre: <Pregame 0x733db170> CLE @ CIN; 12:05PM; TBD vs TBD
DEBUG (16:49:22): Live: <Scoreboard 0x733db170> CLE (3) @ CIN (3); Status: Final: Tied; Inning: (Number: 9; State: End); B:0 S:3 O:3; Bases: [ ][ ][ ]; Notes: 'Game called  with 0 out in the top of the 10th and a 0-0 count on Yu Chang.';
DEBUG (16:49:22): Final: <Final 0x733db170> W: Unknown 0-0 (Unknown); L: Unknown 0-0 (Unknown); S: None (0)

ajbowler, 2019-02-24:
Another one, damn you Spring Training!

twfarley, 2019-02-24:
Might be because it was a tie?

ajbowler, 2019-02-24:
Yes, Tie I think is a new status we haven't accounted for. Need to add it to the list of post-game statuses

ajbowler, 2019-02-24:
Fixed by https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/pull/221

closed 2019-02-24 by ajbowler