MLB-LED-Scoreboard/mlb-led-scoreboard #238: Add commandline arguments as configuration options

pjwarner, 2019-03-15:
For instance I remote into my raspi using my phone often and its tedious to type some of the arguments etc in.  Would be nice to have the command line arguments as a setting in the config.json file

swemoney, 2019-03-16:
I'm unsure if there's a similar app on Android. Someone who uses Android devices would have to chime in on that one. But all of my solutions are available to anyone, but the shortcuts app makes it really easy to just start/stop/restart the with a tap (instead of needing to full on SSH into the scoreboard and type the commands)

ajbowler, 2019-03-17:
To echo what @swemoney said, we opted to keep the LED flags as command line options because:

* The rgbmatrix library owns these flags and uses them via CLI, so we will do the same for consistency
* Only one obvious way to do things for simplicity

If you want to run the scoreboard without having to type in all of the flags you need every time, make an alias. I have an alias on my Pi `scoreboard` which maps to `sudo python ~/mlb-led-scoreboard/main.py --led-gpio-mapping=adafruit-hat --led-cols=64 --led-brightness=60` for example

closed 2019-03-17 by ajbowler