MLB-LED-Scoreboard/mlb-led-scoreboard #245: Crash on news screen.

swemoney, 2019-03-19:
Getting a crash here.

```
Mar 19 00:03:02 mlb-led-scoreboard python[10737]: Traceback (most recent call last):
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "main.py", line 36, in <module>
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     MainRenderer(matrix, data).render()
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 49, in render
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     self.__render_offday()
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 69, in __render_offday
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     self.__update_scrolling_text_pos(renderer.render())
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "/home/pi/mlb-led-scoreboard/renderers/offday.py", line 24, in render
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     text_len = self.__render_news_ticker()
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "/home/pi/mlb-led-scoreboard/renderers/offday.py", line 66, in __render_news_ticker
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     ticker_text = self.data.headlines.ticker_string()
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "/home/pi/mlb-led-scoreboard/data/headlines.py", line 112, in ticker_string
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     ticker = self.__add_string_to_ticker(ticker, countdown_string)
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:   File "/home/pi/mlb-led-scoreboard/data/headlines.py", line 126, in __add_string_to_ticker
Mar 19 00:03:02 mlb-led-scoreboard python[10737]:     return (t + text_to_add)
Mar 19 00:03:02 mlb-led-scoreboard python[10737]: TypeError: cannot concatenate 'str' and 'NoneType' objects
Mar 19 00:03:02 mlb-led-scoreboard systemd[1]: mlb-led-scoreboard.service: Main process exited, code=exited, status=1/FAILURE
Mar 19 00:03:02 mlb-led-scoreboard systemd[1]: mlb-led-scoreboard.service: Unit entered failed state.
Mar 19 00:03:02 mlb-led-scoreboard systemd[1]: mlb-led-scoreboard.service: Failed with result 'exit-code'.
```

msroth, 2019-03-21:
Ditto.
This happened tonight after an upgrade from 3.1.4 -> 3.2.0.

foldedpaper, 2019-03-22:
I see it too, on a fresh install of 3.2.0 (this is my first time setting up the project).

I can work around the issue by either: a) changing `preferred > teams` in `config.json` to a team that plays today, or b) setting `news_ticker > team_offday` to `false`

ajbowler, 2019-03-23:
Fixed in 3.2.1, closed by #252

closed 2019-03-23 by ajbowler