MLB-LED-Scoreboard/mlb-led-scoreboard #246: AttributeError: 'module' object has no attribute 'LAYOUT_PERFECT'

pjwarner, 2019-03-20:
INFO (15:35:51): MLB LED Scoreboard - v3.1.4 (64x32)
Traceback (most recent call last):
  File "main.py", line 36, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 57, in render
    self.__render_game()
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 126, in __render_game
    self.data.refresh_overview()
  File "/home/pi/mlb-led-scoreboard/data/data.py", line 114, in refresh_overview
    self.__update_layout_state()
  File "/home/pi/mlb-led-scoreboard/data/data.py", line 155, in __update_layout_state
    self.config.layout.set_state(layout.LAYOUT_PERFECT)
AttributeError: 'module' object has no attribute 'LAYOUT_PERFECT'

Received the above error earlier today, not 100% sure when but I have to imagine it occurred on or around the Os Red Sox game.  First time I have seen this.

ajbowler, 2019-03-20:
Try updating to 3.2.0, this should have been fixed.

closed 2019-03-20 by ajbowler

pjwarner, 2019-03-20:
Holy fast response...ok thanks!