MLB-LED-Scoreboard/mlb-led-scoreboard #249: Home team background white when teams have same colors

msroth, 2019-03-21:
When teams with the same team colors play (ex.  WSH Nationals and STL Cardinals), the scoreboard shows a big blob of color at the top (in the case of WSH and STL, red).  Is it possible that if teams use the same colors the home team background could be white to give some separation?

swemoney, 2019-03-21:
I don't think this is something we'd do. The accent colors were just released with 3.2.0 and kind of help there a little bit. You can check out the personalization section of the README for information about customizing the colors via your own `ledcolors/colors.json` file.

twfarley, 2019-03-21:
You could also mess with the coords to move down the home team by 1 px and have the default background color poke through.

msroth, 2019-03-21:
Yep, I just got 3.2.0 running and like the accent colors.  Ignore my request.

closed 2019-03-21 by swemoney