MLB-LED-Scoreboard/mlb-led-scoreboard #254: Speed up text scrolling speed

hubster1000, 2019-03-27:


ajbowler, 2019-03-27:
We could certainly try extending the range. At some point you're going to get diminishing returns, based on the performance of your Pi. The faster it gets, the more times your board has to redraw itself to reposition the text

ajbowler, 2019-03-27:
```
SCROLLING_SPEEDS = [0.3, 0.2, 0.1, 0.075, 0.05]
```

This is what we have right now. If 0.05 is too slow for you we could try adding 0.01 as a 5th option and see if that works out. The board may not be fast enough to work with that, though.

davidjmorin, 2020-03-04:
has this been implemented yet?  The speed is terribly slow on pi zero

ajbowler, 2020-03-05:
See the above pull request ^ it's on the latest release so make sure you have the latest.

Pi Zero is not a fast computer. It has far less resources than the other Pis so some extra chug is likely going to happen.

PR #278 (2020-02-18): Scroll speed
Closes #254.

Adds scroll speeds of 0.025 and 0.01 to list of available scroll speeds. Updated debug text for new settings range. Retains default setting of 2.
closed 2020-03-05 by ajbowler