MLB-LED-Scoreboard/mlb-led-scoreboard #265: Error while installing scoreboard

PHI34Halladay, 2019-04-13:
Hey guys,

I installed everything (using the install.sh file) but once I open the main.py file in the terminal I get this error:
[image]

Please help me out here.. I have no clue how to fix this error.

Thanks in advance.

Regards,
Patrick

ajbowler, 2019-04-13:
That looks like a dependency of the matrix library didn't install correctly - it should install for you. How did you run the installation? `sudo ./install.sh` is the proper way to do this. Just doublechecking.

PHI34Halladay, 2019-04-14:
Just did a quick reinstall on my pi. I ran sudo ./install.sh as you said but I get the same error. Does it make any difference wether I've installed python v2.7 oder v3.0?
How do I see that everything is running well? Or in other words, whats the next thing that should happen after running sudo main.py?
Do I necessarily need the led screens connected and running already or can I just try everything before installing the leds?
I'm really thankful for your quick response and willingness to help me out here.

swemoney, 2019-04-14:
We only officially support python 2.7 at this point so that should be fine. If you are getting an error like this on startup, it's clear things aren't running well. When running `sudo ./install.sh` on a fresh clone of the repo, are you waiting for the full thing to finish? There's a part of the install that can take up to 10-20 minutes to finish. Some people get hung up here. You also should have the LED matrix connected to your pi when you run the main scoreboard script but the error you're getting is definitely a dependency that isn't being installed for some reason.

PHI34Halladay, 2019-04-15:
Ok, so the version seems not to be the issue.
I deleted everything that has to do with the scoreboard and did a fresh installation with the same error message. I will probably completely reinstall my raspberry to make sure that it's working.
The installation was finished (at least it said it was finished). It said that a config.json file was created.
I will also connect the led panels to the pi the next time I'll work on this project.
I will keep you guys updated.

Patrick

PHI34Halladay, 2019-04-21:
Hey guys
just wanted to let you know, that I fixed the problem.
I did a complete rebuild of my raspberry and installed a fresh raspian system. After that I ran the installation via putty (didn't do that before) and it worked afterwards.
I don't know if it either was the new installation or just the installation via putty that fixed the issue for me.

Thanks for the support :)

closed 2019-04-21 by ajbowler