MLB-LED-Scoreboard/mlb-led-scoreboard #267: Bad contact

PHI34Halladay, 2019-04-24:
Hey guys,
I finally got the scoreboard to install correctly and it's up and running.
Now I got one final problem, where I'm not sure how to fix it.

[image]
Once I connect those two things I need to manually hold the cable. Otherwise I just get a black screen or best case weird colors running all over the led panel.

Did you guys solder the two ends together in order to get a positive result or is either my adafruit hat or the cable damaged?

Thanks in advance.

swemoney, 2019-04-24:
All of those pieces should be soldered to the hat. We recommend the bonnet if you don't have experience with soldering as it offers the same functionality but is already presoldered for you.

closed 2019-04-24 by swemoney

ajbowler, 2019-04-24:
For support issues pertaining to hardware, you're better off either talking to your vendor (Adafruit in this case), or to the maintainers of the actual LED matrix code https://github.com/hzeller/rpi-rgb-led-matrix

This space is mainly for supporting the scoreboard software.

PHI34Halladay, 2019-04-24:
Ok, sorry @ajbowler  but thanks already @swemoney :)