MLB-LED-Scoreboard/mlb-led-scoreboard #268: 128x64 Error Message

PHI34Halladay, 2019-04-27:
After trying out the scoreboard with two LEDs did I decide to run it using four LEDs.
So I went out and bought two extra 64x32 LED panels.

When I want to run the scoreboard I get this error message:
[image]

I used the w128h64.json and  edited the utils to that:
[image]
I also tried 64x32 together with 4 daisy-chained LEDs. 
The option with two parallels doesn't work either:
[image]

Any help here?

ajbowler, 2019-04-27:
you should not be updating the utils. You should be using the LED flags when running the software, similar to how you supply `adafruit-hat` to the gpio mapping flag.

Have a look through here: https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard#flags

Reset all the changes you've made to the utils code, as this should not be changed.

The command you probably want here is `sudo python main.py --led-gpiomapping="adafruit-hat" --led-cols=128 --led-rows=64`

Add or remove flags as needed to suit your needs.

closed 2019-04-27 by ajbowler

PHI34Halladay, 2019-04-27:
Oh ok, didn't know that - thanks 👍

ajbowler, 2019-04-27:
Facing the side with the LEDs, left to right as far as number goes. The back of the panels typically has an arrow on it indicating the direction. Consult the RGBMatrix README as it might have more info for you.

PHI34Halladay, 2019-04-27:
I believe I am too dumb to get this done.
Could somebody with a 128x64 scoreboard send me a picture with his panel and cable setting?
You're also using the w128h64.json file right?

ajbowler, 2019-04-27:
I am personally not using 128x64, I use 64x32.

Join the slack channel in the README to get faster support on this. For general hardware issues, you are more likely to find support in the RGBMatrix space: https://github.com/hzeller/rpi-rgb-led-matrix