MLB-LED-Scoreboard/mlb-led-scoreboard #269: Trouble running code/python version

dgrochmal, 2019-05-12:
I am having trouble running the code.

I have followed the steps outlined in the Readme. 

After following all of the steps, the command line output gave me the following error `ImportError: No module named mlbgame`

I went to manually install the mlbgame library, and got the following error `AttributeError: 'module' object has no attribute '_main'`

Some googling led me to believe I had to start installing things with python3/pip3.

I got the mlbgame library error to go away, but am now getting:

`Traceback (most recent call last):
  File "main.py", line 2, in <module>
    from data.scoreboard_config import ScoreboardConfig
  File "/home/pi/mlb-scores/mlb-led-scoreboard/data/scoreboard_config.py", line 2, in <module>
    from layout import Layout
ImportError: cannot import name 'Layout'`

Thanks in advance.

ajbowler, 2019-05-12:
You should only be using python 2. Python 3 is not supported by the LED matrix code.

dgrochmal, 2019-05-12:
Thanks for the reply, I’ll give that a try

closed 2019-05-12 by ajbowler

dgrochmal, 2019-05-12:
Got further along, but am still having trouble getting running. Getting the following errors:

`pi@raspberrypi:~/mlb-scores/mlb-led-scoreboard $ sudo python2 main.py --led-gpio-mapping="adafruit-hat" --led-no-hardware-pulse LED_NO_HARDWARE_PULSE
INFO (23:56:49): MLB LED Scoreboard - v3.2.1 (32x32)
Traceback (most recent call last):
  File "main.py", line 36, in <module>
    MainRenderer(matrix, data).render()
  File "/home/pi/mlb-scores/mlb-led-scoreboard/renderers/main.py", line 57, in render
    self.__render_game()
  File "/home/pi/mlb-scores/mlb-led-scoreboard/renderers/main.py", line 88, in __render_game
    self.data.refresh_overview()
  File "/home/pi/mlb-scores/mlb-led-scoreboard/data/data.py", line 113, in refresh_overview
    self.overview = mlbgame.overview(self.current_game().game_id)
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/__init__.py", line 198, in overview
    return mlbgame.game.Overview(mlbgame.game.overview(game_id))
  File "/usr/local/lib/python2.7/dist-packages/mlbgame/game.py", line 328, in overview
    raw_box_score_root = etree.parse(raw_box_score).getroot()  File "src/lxml/etree.pyx", line 3435, in lxml.etree.parse
  File "src/lxml/parser.pxi", line 1861, in lxml.etree._parseDocument
  File "src/lxml/parser.pxi", line 1881, in lxml.etree._parseFilelikeDocument
  File "src/lxml/parser.pxi", line 1776, in lxml.etree._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 1187, in lxml.etree._BaseParser._parseDocFromFilelike
  File "src/lxml/parser.pxi", line 601, in lxml.etree._ParserContext._handleParseResultDoc
  File "src/lxml/parser.pxi", line 711, in lxml.etree._handleParseResult
  File "src/lxml/parser.pxi", line 640, in lxml.etree._raiseParseError
  File "http://gd2.mlb.com/components/game/mlb/year_2019/month_05/day_11/gid_2019_05_11_wasmlb_lanmlb_1/rawboxscore.xml", line 1
lxml.etree.XMLSyntaxError: Space required after the Public Identifier, line 1, column 55`