MLB-LED-Scoreboard/mlb-led-scoreboard #273: Rotate between division standings and completed/upcoming games

NovaCyntax, 2019-08-18:
When preferred team's game ends:
Rotate through selected division standings once
Rotate through completed/upcoming games once
Repeat alternating rotations

Was unable to figure out how to do this with the provided config options. Thanks for all the work you guys put in, this is a neat project and I'm enjoying it. Considering expanding it to a 64x64 display at some point in the future so I can contribute a layout

swemoney, 2019-08-18:
This is definitely a thing I've had planned for a bit now but it still does require a bit of work. It will definitely be very configurable when it's finally done.

NovaCyntax, 2019-08-18:
Appreciate the speedy response, if you need a tester let me know

closed 2022-04-13 by twfarley