MLB-LED-Scoreboard/mlb-led-scoreboard #283: Does not complete install - MLBGAME missing

davidjmorin, 2020-03-04:
I followed the instructions running this on a rasp pi zero w headless.

Hey all. My Pi keeps getting stuck at the part when it is building wheels for collected packages: mlbgame, lxml

It just hangs and nothing ever happens.

Im installing the git repository http://github.com/MLB-LED-SCOREBOARD/mlb-led-scoreboard

Has anyone had this happen? What is a fix for this?

I am running rasbian lite on rasp pi zero w

Install Log: https://pastebin.com/aLVgp0bH

I try to manually install mlbgame and this is where it gets stuck

```
pi@raspberrypi:~/mlb-led-scoreboard $ sudo python main.py --led-gpio-mapping=adafruit-hat --led-brightness=60 --led-slowdown-gpio=2
Traceback (most recent call last):
  File "main.py", line 3, in <module>
    from renderers.main import MainRenderer
  File "/home/pi/mlb-led-scoreboard/renderers/main.py", line 1, in <module>
    from data.final import Final
  File "/home/pi/mlb-led-scoreboard/data/final.py", line 1, in <module>
    import mlbgame
ImportError: No module named mlbgame
pi@raspberrypi:~/mlb-led-scoreboard $ sudo pip install mlbgame
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting mlbgame
  Downloading https://files.pythonhosted.org/packages/e8/f2/0ee0ea8de91acda582766bb86ebb358a84670006390509e30df2dd1b90bf/mlbgame-2.5.0.tar.gz
Collecting lxml (from mlbgame)
  Using cached https://files.pythonhosted.org/packages/39/2b/0a66d5436f237aff76b91e68b4d8c041d145ad0a2cdeefe2c42f76ba2857/lxml-4.5.0.tar.gz
Building wheels for collected packages: mlbgame, lxml
  Running setup.py bdist_wheel for mlbgame ... done
  Stored in directory: /root/.cache/pip/wheels/61/42/df/ed3391d4f690b4b034d18744820172b2c5d5809d55f03b2409
  Running setup.py bdist_wheel for lxml ... -

```

twfarley, 2020-03-04:
It's been suggested that the pi zero doesn't have enough memory to complete that task. See this Slack thread: https://mlb-led-scoreboard.slack.com/archives/C9BLJ9JF9/p1577157789002400

davidjmorin, 2020-03-04:
I got it to work but i had to run 2 commands
sudo apt-get install libxml2-dev libxslt-dev python-dev
and
sudo apt-get install python3-lxml python-lxml

The pip install mlbgame

Works now.  Maybe an updae needs to happen to this git?

pawptart, 2020-03-05:
There was an issue last year (#215) that came up and the scoreboard now uses a fork of [mlbgame](https://github.com/ajbowler/mlbgame).

Could potentially be related?

Also just sounds like `lxml` failed to install, possibly due to memory constraints like @twfarley suggested. It's been a known issue before (#277)

ajbowler, 2020-03-05:
Us using a fork is a separate thing and not related to the installation slowness on PiZ. I truly think this is just something too memory intensive for a smaller Pi. Even on my Pi 3 the lxml installation takes quite a bit of time

ajbowler, 2020-03-05:
@davidjmorin I don't think `python3-lxml` would be necessary. The scoreboard software was built with Python 2 in mind because the underlying matrix library is written with Python 2 in mind.

Still unsure what true root cause for slowness is if it's not just a hardware resourcing issue.

trevor-viljoen, 2020-03-05:
You can also build the matrix bindings for Python3: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python

```sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
make build-python PYTHON=$(which python3)
sudo make install-python PYTHON=$(which python3)```

davidjmorin, 2020-03-05:
Yea I am not sure either.  It installed fine when i ran them separate.  Then I ran the ./install.sh again and everything worked.  Took about 10 min to complete but now i have a working board.

ajbowler, 2020-03-05:
> You can also build the matrix bindings for Python3: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python
> 
> ```
> sudo apt-get update && sudo apt-get install python3-dev python3-pillow -y
> make build-python PYTHON=$(which python3)
> sudo make install-python PYTHON=$(which python3)
> ```

Yeah it's doable, but documented to be slower - https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python#performance

I haven't personally tested it so I don't have actual benchmarks.

closed 2020-03-05 by ajbowler