MLB-LED-Scoreboard/mlb-led-scoreboard #285: Level of effort to display two games at once?

ryanhagerty, 2020-03-07:
I was working on a custom layout to optimize the screen real estate and was wondering if it's possible to display two games at once?

[image]

From what I can tell, it isn't possible from a purely config standpoint unless I'm mistaken.

ajbowler, 2020-03-07:
Yeah this would be a breaking change to the configs and layouts. Level of effort I think logically wouldn't be too bad as we'd just need to essentially create two scoreboard objects to draw at a time, most of the effort would be in the rendering and positioning.

ajbowler, 2020-03-07:
I don't know that I'll personally get to this or other major scoreboard features anytime soon but as always with this project I encourage the community to help contribute

long-horse, 2020-03-20:
What I would love is to somehow run an MLB game and an NHL game. Maybe even 2 matrices, one for each. 2 games of each sport would be next level :o

ajbowler, 2020-03-20:
You would need a separate project/application entirely to do that, the two boards are two different software.

pawptart, 2020-03-22:
@long-horse There's a script out there to swap between MLB and NHL:

https://github.com/necromancyr/swap-led-scoreboard

long-horse, 2020-03-23:
@pawptart Thanks for that, although I have no idea how to implement it and there is minimal documentation :| It seems like it has a few dependencies that I don't know how to install, namely the "supervisor properly setup to run ALL the boards you want run". I also am not familiar with gpiozero. And it sounds like I need a button too.

pawptart, 2020-03-23:
https://github.com/pawptart/swap-led-scoreboard/tree/buttonless

Try this. I haven't tested it but it strips out all the stuff with the buttons that you won't need. Also there's a good writeup of how to use Supervisor in the NHL board repo (that's apparently what inspired this script).

long-horse, 2020-03-24:
I figured out supervisor.

Iassume I need to `git clone` your project? 

And I most likely will add a button at some point, so would I rm -Rf this dir and clone necromancyr's at the point in time?

To run your script, I am guessing I cd into the directory and do `python swapscript.py`, but how do I make it auto run? Or is the pi just supposed to stay on?

closed 2022-04-13 by twfarley