MLB-LED-Scoreboard/mlb-led-scoreboard #287: No live pre-season games?

msroth, 2020-03-08:
I'm noticing for preseason games I am only getting the scores from the day before, and not the live game.  GameDay is showing the live game so I know it is going on and there is a data feed.  Are the XML files not being updated anymore now that the API is REST-based?

ryanhagerty, 2020-03-08:
Hey - just chiming in to say I am able to see live pre-season games. I have noticed that sometimes there are artifacts where a previous day's game will be listed as 'Warm up' or 'Final'. 

I think this might be related to the MLB data library that's being used. I remember the author saying that spring training game data is unreliable.

ajbowler, 2020-03-09:
Spring training data is indeed very unreliable at times. I also see live games but it's not always the best. Often times the pitches/balls/strikes aren't perfect, at least that's what I notice the most.

msroth, 2020-03-09:
Thanks for the responses, I'll keep a closer eye on it and see if I get any real-time data in the next few weeks.  I tried to plumb the MLB XML data that I thought the data library was using and it doesn't seem to be accessible any more.  Has everything moved to the REST API?  I didn't dig any deeper.

trevor-viljoen, 2020-03-09:
@msroth They just disabled the ability to crawl the data without knowing specific URLs. The data is still there: http://gd2.mlb.com/components/game/mlb/year_2020/month_03/day_09/scoreboard.xml

closed 2020-03-09 by ajbowler

msroth, 2020-03-10:
Ah, thanks.  That wasn't very nice of them.