MLB-LED-Scoreboard/mlb-led-scoreboard #288: Matrix won't display correctly

cptdrummer, 2020-03-19:
So I realize there aren't even games right now, but I've been trying to get this set up for the first time without much success.

I'm using a Pi ZeroW, with an Adafruit Bonnet and a 32x64 (rows x cols) LED matrix. I was able to install everything with no errors following the additional suggestions in issue #283 
I've got a 5V 8A power supply plugged into the bonnet to power the martix, a power supply plugged into the Pi, the data cable connecting the bonnet and power from the bonnet to the matrix.

My issue with the LED matrix occurs running both main.py and any of the demos, with similar, erratic behavior. 
I'll provide an example of a demo attempt.

I submit the following command through putty from my laptop, and seem to get normal outputs:
[image]

However, while running, nothing initially displays on the matrix. If I walk over to it, and just wave my hand around near the bonnet -- without touching anything -- this happens:
[image]
(Apologies for the over exposed photo, I wanted to take a video but didn't realize github won't let me upload one)
Random colors flash on and off, in no discernible pattern until I escape the command.

Has any one seen behavior like this before? Any idea what the issue might be? 

Everything with the install appears to be ok, the Pi behaves normally, and everything clearly has power. Short of making a silly syntax error when calling the command, my only other thought is I did a poor job soldering the pins onto the Pi. It won't let me upload an image of my soldering work, so I'll try to include it in a comment. But the soldering looks fairly clean and I can't see any evidence that I shorted any pins.

So I'm pretty stumped here and would appreciate any advice. Hopefully I'm just being stupid. Thanks so much for the help!

cptdrummer, 2020-03-19:
Here is the soldering. Definitely not perfect and I could improve some of the contacts but it definitely looks like each pin has at least a good enough contact 
[image]

long-horse, 2020-03-20:
I'm in the same situation as you (just getting started with this).

First, unrelated, you don't need USB power to the pi with the bonnet power ;) Unless you don't mind the extra cable.

Next, it looks like youre running the matrix demo from the mlb directory.

try `cd ..` until youre at the top level and then `cd rpi-rgb-led-matrix/examples-api-use; sudo ./demo -D11 --led-rows=32 --led-cols=64`

cptdrummer, 2020-03-20:
I'll definitely check out going without USB power, but for now I think I'll take it one thing at a time.

I didn't have the `rpi-rgb-led-matrix` directory, so I cloned it from here - https://github.com/hzeller/rpi-rgb-led-matrix/ 
Maybe the fact that I had to add and make that directory is indicative that something went wrong with my install? 

But either way, switching to the `rpi-rgb-led-matrix/examples-api-use` directory and running your recommended command: `sudo ./demo -D11 --led-rows=32 --led-cols=64` resulted in more of the same behavior:
https://gfycat.com/delectablesmugcarp

With regard to your issue, have you tried running with the `--led-no-hardware-pulse` flag? My understanding is that it should at least help fix it as a troubleshooting step. If that fixes the problem, you'll have to turn off on-board sound, as detailed here: https://github.com/hzeller/rpi-rgb-led-matrix/#Troubleshooting

pawptart, 2020-03-22:
> Here is the soldering. Definitely not perfect and I could improve some of the contacts but it definitely looks like each pin has at least a good enough contact

Yikes man, that soldering is no good. You need to get your iron hotter and reflow the solder:

[image]

All of your joints look like the A joint. It needs to be shiny and not globby, and ideally you won't see the copper pad underneath the pin.

long-horse, 2020-03-23:
> With regard to your issue, have you tried running with the `--led-no-hardware-pulse` flag? My understanding is that it should at least help fix it as a troubleshooting step. If that fixes the [...]

`--led-no-hardware-pulse` does nto seemed to have helped so I think I may need the hardware mod soldering a wire between 4 & 18.

cptdrummer, 2020-03-23:
Well, either I completely forgot how to correctly solder since leaving college, or I never learned how to do it right in the first place. Either way, after an incredibly enlightening 3 minute youtube tutorial, it looks like I've got my fix. Running the examples results in a well behaved display. 

https://gfycat.com/talldeficientfrogmouth

Thanks @pawptart for the heads up about the shoddy workmanship.

closed 2020-03-23 by cptdrummer