MLB-LED-Scoreboard/mlb-led-scoreboard #293: MLB XML API deprecation

pawptart, 2020-07-25:
It appears that MLB no longer supports its older GameDay XML API. The API wrapper used by the scoreboard [`mlbgame`](https://github.com/panzarino/mlbgame) depends on this API to render team standings.

There have been lots of reports of the scoreboard warning on startup `Failed to refresh standings.` due to this issue.

A sample API call to retrieve stats:
```
// URL
GET http://mlb.mlb.com/lookup/json/named.historical_standings_schedule_date.bam?season=2020&game_date=%272020/07/25%27&sit_code=%27h0%27&league_id=103&league_id=104&all_star_sw=%27N%27&version=48

// Response
'Check back soon!  Working on you request'
```

Short term, an appropriate fix may be to simply bypass the `mlbgame` dependency/old API only on the broken call and use the new [`MLB Data API`](https://appac.github.io/mlb-data-api-doc). 

Sample call to the new endpoint:

```
// URL
http://lookup-service-prod.mlb.com/json/named.team_all_season.bam?sport_code='mlb'&all_star_sw='N'&sort_order=name_asc&season='2020'

{
    "team_all_season": {
        "copyRight": " Copyright 2020 MLB Advanced Media, L.P.  Use of any content on this page acknowledges agreement to the terms posted here http://gdx.mlb.com/components/copyright.txt  ",
        "queryResults": {
            "created": "2020-07-25T18:38:43",
            "totalSize": "30",
            "row": [
                {
                    "phone_number": "(602) 462-6500",
                    "venue_name": "Chase Field",
                    "franchise_code": "ARI",
                    "all_star_sw": "N",
                    "sport_code": "mlb",
                    "address_city": "Phoenix",
                    "city": "Phoenix",
                    "name_display_full": "Arizona Diamondbacks",
                    "spring_league_abbrev": "CL",
                    "time_zone_alt": "America/Phoenix",
                    "sport_id": "1",
                    "venue_id": "15",
                    "mlb_org_id": "109",
                    "time_zone_generic": "MST",
                    "mlb_org": "Arizona Diamondbacks",
                    "last_year_of_play": "2021",
                    "league_full": "National League",
                    "home_opener_time": "9:40:00 PM",
                    "address_province": "",
                    "league_id": "104",
                    "name_abbrev": "ARI",
                    "bis_team_code": "ARI",
                    "league": "NL",
                    "spring_league": "CL",
                    "base_url": "dbacks.mlb.com",
                    "address_zip": "85004",
                    "sport_code_display": "Major League Baseball",
                    "mlb_org_short": "Arizona",
                    "time_zone": "MST",
                    "address_line1": "401 East Jefferson Street",
                    "mlb_org_brief": "D-backs",
                    "address_line2": "",
                    "season": "2020",
                    "address_line3": "",
                    "division_abbrev": "NLW",
                    "name_display_short": "Arizona",
                    "team_id": "109",
                    "active_sw": "Y",
                    "address_intl": "N",
                    "state": "AZ",
                    "address_country": "",
                    "mlb_org_abbrev": "ARI",
                    "division": "W",
                    "team_code": "ari",
                    "name": "D-backs",
                    "website_url": "dbacks.com",
                    "sport_code_name": "MLB",
                    "first_year_of_play": "1996",
                    "league_abbrev": "NL",
                    "name_display_long": "Arizona Diamondbacks",
                    "store_url": "arizona.diamondbacks.mlb.com",
                    "time_zone_text": "MST",
                    "name_short": "Arizona",
                    "home_opener": "2020-07-30T00:00:00",
                    "address_state": "AZ",
                    "division_full": "National League West",
                    "time_zone_num": "-7",
                    "spring_league_full": "Cactus League",
                    "address": "401 East Jefferson Street&#xa;Phoenix, AZ&#x9;85004",
                    "name_display_brief": "D-backs",
                    "file_code": "ari",
                    "division_id": "203",
                    "spring_league_id": "114",
                    "venue_short": "Chase Field"
                 },
             
                 . . . More teams . . .
            ]
        }
    }
}
```

Long term it is most likely best to get rid of the `mlbgame` dependency and start using the appopriate API.

pfightingpolish, 2020-07-25:
I was able to Frankenstein the [`MLB-StatsAPI`](https://github.com/toddrob99/MLB-StatsAPI) to make the standings work again. 

My code, as it has been throughout my journey, was an inelegant solution, so I'm not even going to try and share it in full. I've been heavily customizing my board and my Python is weak. I wasn't sure how to iterate through the teams and I've been lazy about coding back to the config.json file for the coordinates. I just hard-coded out the data five times, once for each position in the division, with mostly constants for the coordinates. Here's what first place looks like, if it helps anyone: 

```
      team_text = self.__team_shortener(self.data.current_standings().get('teams')[0].get('name'))
      graphics.DrawText(self.canvas, font["font"], 2, 22, self.team_name_color, team_text)

      record_text = "{}-{}".format(self.data.current_standings().get('teams')[0].get('w'), self.data.current_standings($
      if "-" in self.data.current_standings().get('teams')[0].get('gb'):
        gb_text = " -  "
      else:
        gb_text = format(self.data.current_standings().get('teams')[0].get('gb'))

      graphics.DrawText(self.canvas, font["font"], record_start, 22, self.team_stat_color, record_text)
      graphics.DrawText(self.canvas, font["font"], gb_start, 22, self.team_stat_color, gb_text)
```

`__team_shortener` is just a translation function that takes the full team name ("Houston Astros") and translates it to just the city "Houston"); I'm using a 64x128 board with a variable width font, so I can fit that on there as long as the Giants don't win or lose 100 games (which ain't happening this year either way). `get` is a powerful tool in the [`MLB-StatsAPI`](https://github.com/toddrob99/MLB-StatsAPI). Changing `[0]` to 1, 2, 3 and 4 will give you the other places. 

Standings is a good gateway to [`MLB-StatsAPI`](https://github.com/toddrob99/MLB-StatsAPI), as the standings work somewhat separately from the rest of the board. However, I'm going to warn that moving from [`mlbgame`](https://github.com/panzarino/mlbgame) to [`MLB-StatsAPI`](https://github.com/toddrob99/MLB-StatsAPI) appears to be a massive task that will require an almost complete redo of the code. I started poking around to try and see if I could get the current batter & pitcher names onto the scoreboard, which is very possible in the new API: 

```
print statsapi.get('game_linescore',{'gamePk':631642}).get('offense').get('batter')
print statsapi.get('game_linescore',{'gamePk':631642}).get('defense').get('pitcher').get('fullName')
```

But I realized quickly it wasn't going to be easy. The big thing is the difference in the key field used: [`mlbgame`](https://github.com/panzarino/mlbgame) works off the old game ID format (i.e. gid_2011_07_09_tbamlb_nyamlb_1) whereas the [`MLB-StatsAPI`](https://github.com/toddrob99/MLB-StatsAPI) works off the new one, where the ID is just an integer (631642 above). A healthy amount of the [`mlb-led-scoreboard`](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard) code, at present, routinely splits up the old game ID and pulls info based off, say, the team abbreviation. The new one's going to require different translation in a bunch of places. I gave up on trying to translate the game ID to the gamePk being sought above. 

I'm not optimistic about any interoperability between the two, as it doesn't appear [`mlbgame`](https://github.com/panzarino/mlbgame) has been touched in about two years and has unanswered issues. I think we're more apt to just get `mlb-led-scoreboard` upgraded to 4.0. 

That said ... I was actually considering coming here and saying **THANK YOU, THANK YOU, THANK YOU** to all of you who have worked on this. I've had a fascination with scoreboards since the first time I went to a Brewers game as a five-year-old in 1987. This code has allowed me to recapture some of the awe and wonder. I've been obsessed from the moment I saw the Dodger example on Reddit. 

This code has had me trying to recreate the feeling of that old County Stadium board, familiar to anyone who saw the classic film "Major League" (made in Milwaukee). That's meant some custom fonts, really messing with the code and some late nights. But the results are ... pretty. 

[image]

[image]

[image]

[image]

[image]

[image]

Having taken the code about as far as I think I can, the next task is the carpentry to build a housing. You can guess what I've got in mind.

msroth, 2020-07-26:
Love your classic scoreboards.  Too bad the MLB-LED scoreboard code is broken due the discontinued XML API.  I just installed one of these in a really nice box in a friend's bar.  Imagine our disappointment.  At least the NHL scoreboard will work next month.

pfightingpolish, 2020-07-26:
Games and headlines are still working. It's just the standings that are borked at the moment.

pawptart, 2020-07-26:
> Love your classic scoreboards. Too bad the MLB-LED scoreboard code is broken due the discontinued XML API. I just installed one of these in a really nice box in a friend's bar. Imagine our disappointment. At least the NHL scoreboard will work next month.

Most of the XML API is still working, so anything that drives the actual game stats is still good for. It's just the standings API has apparently been migrated to StatsAPI. The fact that standings don't work doesn't break anything, but it is less interesting on off days.

pawptart, 2020-07-26:
> However, I'm going to warn that moving from [`mlbgame`](https://github.com/panzarino/mlbgame) to [`MLB-StatsAPI`](https://github.com/toddrob99/MLB-StatsAPI) appears to be a massive task that will require an almost complete redo of the code.

Absolutely true. My understanding is the XML API is going away completely at some indeterminate point in the future.

swemoney, 2020-07-26:
There was work being done last year to kind of get our own mlbgame library running using the json API. I haven't had a lot of time lately to take a look but I'll try and see what's working and what still needs work. The idea is to ultimately ditch the mlbgame library in favor of one that supports the json API. Just a question of if we have to write it ourselves or not.

pawptart, 2020-07-26:
Just a cursory look at the codebase, seems like there are only a few API calls made through `mlbgame`:

```python
# The call this issue concerns
mlbgame.standings()

# Calls to grab today's game data
mlbgame.day()

# Not exactly sure what this returns, there may not be an analog for this on StatsAPI
mlbgame.overview()
```

There may be some child calls that I'm missing by a first pass but based on that I don't think writing our own library to support the JSON API is a bad idea.

pfightingpolish, 2020-07-26:
Overview pulls in a lot of the info specific to a particular game, like team runs, hits, errors, the count and outs, etc. (though not current batter and pitcher, frustratingly). 

```
class Overview(mlbgame.object.Object):
    """Object to hold an overview of game information
    Properties:
        ampm
        attendance
        aw_lg_ampm
        away_ampm
        away_code
        away_division
        away_file_code
        away_games_back
        away_games_back_wildcard
        away_league_id
        away_loss
        away_name_abbrev
        away_preview_link
        away_probable_pitcher_era
        away_probable_pitcher_first
        away_probable_pitcher_first_name
        away_probable_pitcher_id
        away_probable_pitcher_last
        away_probable_pitcher_last_name
        away_probable_pitcher_losses
        away_probable_pitcher_name_display_roster
        away_probable_pitcher_number
        away_probable_pitcher_s_era
        away_probable_pitcher_s_losses
        away_probable_pitcher_s_wins
        away_probable_pitcher_stats_season
        away_probable_pitcher_stats_type
        away_probable_pitcher_throwinghand
        away_probable_pitcher_wins
        away_recap_link
        away_sport_code
        away_team_city
        away_team_errors
        away_team_hits
        away_team_id
        away_team_name
        away_team_runs
        away_time
        away_time_zone
        away_win
        balls
        date
        day
        double_header_sw
        elapsed_time
        first_pitch_et
        game_data_directory
        game_id
        game_nbr
        game_pk
        game_type
        gameday_link
        gameday_sw
        hm_lg_ampm
        home_ampm
        home_code
        home_division
        home_file_code
        home_games_back
        home_games_back_wildcard
        home_league_id
        home_loss
        home_name_abbrev
        home_preview_link
[... 100 lines cut ...]
>>> i = 0
>>> while i < 15:
...  statsapi.schedule("2020-07-25")[i].get('game_id')
...  i += 1
...
631642
631666
631515
630992
630967
631108
630942
631241
631181
631453
631382
631565
631603
630899
631092
```

However, you need to somehow determine how many games are in a day, otherwise it throws an out of range error if you try to pull on a day when less than the whole league is playing. And, as you can see, it's not like game IDs have any rhyme or reason. 

It's possible to make this work, and even open up more possibilities for data to be shown on the board. It's late and I've literally only been working in Python for a few days, so I'm not going to solve all the riddles at the moment. But, maybe this'll help the folks working on making it happen.

trevor-viljoen, 2020-07-26:
I think I actually wrote that section of code for mlbgame to pull the standings. I haven’t had time to dig into the new issue, but I did some work on writing python3 bindings for the json API a while back. Feel free to dig through the code, fork, make pull requests or suggestions. 

https://github.com/trevor-viljoen/mlbapi

msroth, 2020-07-26:
> Most of the XML API is still working, so anything that drives the actual game stats is still good for. It's just the standings API has apparently been migrated to StatsAPI. The fact that standings [...]

This hasn't been my experience.  I get nothing but day old game results.  No live game data at all.  I'm going to mess with my config, though nothing has changed since last season.

pawptart, 2020-07-26:
> ```
> >>> i = 0
> >>> while i < 15:
> ...  statsapi.schedule("2020-07-25")[i].get('game_id')
> ...  i += 1
> ```

Just FYI, the Pythonic way to do this is a `for ... in` loop:
```
return [game.get('game_id') for game in statsapi.schedule("2020-07-25")]
```

Conveniently avoids having to hard code the number of teams.

pfightingpolish, 2020-07-26:
> This hasn't been my experience. I get nothing but day old game results. No live game data at all. I'm going to mess with my config, though nothing has changed since last season.

Weird. Mine's chugging along just fine right now. 

Things I can think of that might cause old results: 

- Having Demo Date set to anything other than false in `config.json`
- Something's up with your day start time in `config.json`
- Your Pi could use a restart
- Your Pi has networking issues

Lots of other possibilities, too, but those are the first I can think of. 

Best of luck troubleshooting.

pfightingpolish, 2020-07-26:
> Just FYI, the Pythonic way to do this is a `for ... in` loop:
> 
> ```
> return [game.get('game_id') for game in statsapi.schedule("2020-07-25")]
> ```
> 
> Conveniently avoids having to hard code the number of teams.

Yeah, I'm still getting my Pi sea legs. I will not be the expert coder here. Totally makes sense that way.

msroth, 2020-07-26:
> Weird. Mine's chugging along just fine right now.
>
> Things I can think of that might cause old results:
>
> * Having Demo Date set to anything other than false in `config.json`
> * Something's up with [...]

Thanks for the confirmation that games are still working and for the troubleshooting advice.  I had tried all those things except the "end of day" setting.  When I compared that section of the config file to the one in GitHub, I noticed there is a  time_format setting and the end of day is now "00:00".  Once those were changed it worked!  Thanks a million.

pfightingpolish, 2020-08-18:
Just because there's no better place to post this, here's my finished product. 

It has over 1,000 likes in a 50,000+ Facebook group for Milwaukee nostalgia in just one day. I'm friends with a guy who worked with the real thing for many years, and he said both that he's impressed with it and that he's hoping to get a picture put in the control room for Miller Park's scoreboard (I am really tempted to build one for them, too). 

Thank you all so much for working on this. This thing feels like it's now my most prized possession. 

[image]

[image]

twfarley, 2020-08-18:
This is awesome! Are you planning to check in your changes to a fork?

pfightingpolish, 2020-08-18:
I'd be willing to. I'd have to figure out how to do it. This is literally the first thing I've done involving Github. 

As I've said upthread, too, I will warn my code is, admittedly, somewhat disastrous. To get some of my changes, I translated [mlbgame](https://github.com/panzarino/mlbgame)'s game IDs into something that can pull the game IDs from [MLB-StatsAPI](https://github.com/toddrob99/MLB-StatsAPI), but to say my solutions were inelegant is putting it mildly. There were more than a few places where I probably should have referenced back to the `config.json` file, coordinates file or color file, but just hard-coded values in the interest of speed. My CS252 teacher from college would probably try and snap a ruler over my head for some of my code. 

Nonetheless, just like with my finagling the code, I can try and figure it out. I have some screen fatigue with all the social media hype right now, but I will remain active in this community and will try and create a fork when I get a chance.

pfightingpolish, 2020-08-29:
> This is awesome! Are you planning to check in your changes to a fork?

I just created a fork to my profile and uploaded the files I changed. 

A quick outline of some of my differences. You would definitely not be able to run my code out of the box, but can certainly consult for some changes: 

- All the fonts are custom and variations on what was actually used on the County Stadium Omega board. Because the matrix directory on github is a direct link to hzeller's, I have my variant fonts in my main directory rather than in the matrix directory, where I have them housed on my install. They're just Fony modifications and the numbered sizes don't necessarily correspond to anything, as they're variable width.
 
- The whole board is in old-school amber. My install was 128x64 and gave a decent amount of flicker with the color, so I decided to go single-color for the whole thing. Take that into account. 

- Obviously the layout is different, but I think the big addition I made was to add current batter and pitcher to in-progress games. 

Yes, that pulls info from [MLBStats-API](https://github.com/toddrob99/MLB-StatsAPI), so there's also code to translate the game ID from [mlbgame](https://github.com/panzarino/mlbgame) to the gamePk in [MLBStats-API](https://github.com/toddrob99/MLB-StatsAPI). I've been poking around [MLBStats-API](https://github.com/toddrob99/MLB-StatsAPI) more, and I think I might have done this a much harder way than I had to, as it looks like there is a version of the old ID in the new 'game' endpoint. But there's also so much in `game` that I think it might be better and more efficient to narrow it down. 

I've started looking at trying to get play results to scroll, too, after they occur, but if I do that, I'd probably go through the `game-playByPlay` endpoint separately from the `game-linescore` endpoint where I get batter and pitcher rather than storing the general `game` endpoint, as then, you're storing every bit of StatCast data in a game — hot zones, pitchF/X, you name it — which might be too much for the Pi. 

- I also made the point to show a linescore with RHE (runs, hits and errors) rather than just each team and score. That still uses data from the old API, however. Probably only possible on w128 installs, but I think it's a useful addition. With that information on the board, I virtually turned off any no-hitter/perfect-game notifications by setting the number of innings before they display up to 20 (and if a team ever gets no-hit for 20 innings in a single game, I'll accept the scoreboard crashing). 

- I also fixed the standings by pulling from [MLBStats-API](https://github.com/toddrob99/MLB-StatsAPI). My ticker screen is personalized with different layout, team logo and MLB logo. I have mine set up for offdays to run the ticker, but cycle through the standings every five minutes. 

- I left a few of my "party trick" files in there, namely any of the files labeled `offdayXXXXX.py`. Those go to show that you can use your display to show other things, too. I will move around the offday files in the directory to do different stuff, like recreate the board from Rick Vaughan's entrance near the end of _Major League_ and posting pictures of friends with their name and stuff. Typically, I turn on the flag in `config.json` that makes it show the offday screen, then move the file that's doing what I want it to do into that spot rather than the ticker file, usually renaming the ticker file something like `offdayHOLD.py` while I'm doing my other schtuff. It gives you the chance to do some neat things that aren't out of line with some of the other, non-scoreboard things you can do with a matrix. 

Anyhow, check it out if you want to see some of the ways you can customize your install. 

I actually had a reporter from our local CBS affiliate in town over the other day to do a story on my board that should air in a couple weeks. I cited Github a bunch of times; we'll see if it makes the final cut.

WardBrian, 2021-06-16:
Has there been any progress on this? I’m now getting errors on the non-existence of rawboxscore.xml from mlbgame, so I think it’s edged closer to the point of the XML api being completely defunct

WardBrian, 2021-06-18:
I've re-written the data and standings to both use `statsapi` in my fork here: https://github.com/WardBrian/mlb-led-scoreboard/tree/stats-change

I've also made a bunch of other changes (including using py3, putting at-bat information in a similar way to pfightingpolish, and a few other items), so I'm not sure if it would work as a PR, but if anyone is curious it's available

ty-porter, 2022-04-08:
Closed by #335

closed 2022-04-08 by ty-porter