MLB-LED-Scoreboard/mlb-led-scoreboard #297: New installation gives "KeyError: 'preferred"' on run

t0xicchrome, 2020-08-27:
I tried running the main.py but it always seems to error out. Almost as if it's not able to read the config.json. I can confirm that the config.json that was created by the setup is in the directory with read permissions. I am kinda stumped on this one.

Here is the error at runtime:

```
root@raspberrypi:~/mlb-led-scoreboard# sudo python main.py --led-gpio-mapping=adafruit-hat-pwm --led-cols=64 --led-rows=32 --led-chain=2 --led-brightness=60
INFO (19:34:02): MLB LED Scoreboard - v4.0.0 (128x32)
Traceback (most recent call last):
  File "main.py", line 29, in <module>
    config = ScoreboardConfig("config", matrix.width, matrix.height)
  File "/root/mlb-led-scoreboard/data/scoreboard_config.py", line 22, in __init__
KeyError: 'preferred'
```

I ran the install again to see if it missed something, here is that log:

```
root@raspberrypi:~/mlb-led-scoreboard# ./install.sh
Running rgbmatrix installation...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Fetched 15.0 kB in 1s (11.8 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'python-pil' instead of 'python-pillow'
python-pil is already the newest version (4.0.0-4+deb9u2).
python2.7-dev is already the newest version (2.7.13-2+deb9u4).
0 upgraded, 0 newly installed, 0 to remove and 118 not upgraded.
make -C ./lib
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/root/mlb-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/root/mlb-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/root/mlb-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/root/mlb-led-scoreboard/matrix/lib'
python setup.py build --build-lib .
running build
running build_py
running build_ext
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/bindings/python'
make -C ./lib
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/lib'
make -C examples-api-use
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/examples-api-use'
make -C ../lib
make[2]: Entering directory '/root/mlb-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/root/mlb-led-scoreboard/matrix/lib'
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/examples-api-use'
make -C bindings/python build
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/bindings/python'
make -C ../../lib
make[2]: Entering directory '/root/mlb-led-scoreboard/matrix/lib'
make[2]: Leaving directory '/root/mlb-led-scoreboard/matrix/lib'
python setup.py build --build-lib .
running build
running build_py
running build_ext
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/bindings/python'
make -C bindings/python install
make[1]: Entering directory '/root/mlb-led-scoreboard/matrix/bindings/python'
python setup.py install
running install
running build
running build_py
running build_ext
running install_lib
running install_egg_info
Writing /usr/local/lib/python2.7/dist-packages/rgbmatrix-0.0.1.egg-info
make[1]: Leaving directory '/root/mlb-led-scoreboard/matrix/bindings/python'
Obtaining file:///root/mlb-led-scoreboard/matrix/bindings/python
Installing collected packages: rgbmatrix
  Found existing installation: rgbmatrix 0.0.1
    DEPRECATION: Uninstalling a distutils installed project (rgbmatrix) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling rgbmatrix-0.0.1:
      Successfully uninstalled rgbmatrix-0.0.1
  Running setup.py develop for rgbmatrix
Successfully installed rgbmatrix
Installing required dependencies. This may take some time (10-20 minutes-ish)...
HEAD is now at 73f2be8 Bump version
M       matrix
Already on 'master'
Your branch is up-to-date with 'origin/master'.
Already up-to-date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libxslt1-dev' instead of 'libxslt-dev'
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2+deb9u2).
libxslt1-dev is already the newest version (1.1.29-2.1+deb9u2).
0 upgraded, 0 newly installed, 0 to remove and 118 not upgraded.
Requirement already satisfied: pytz in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: tzlocal in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: feedparser in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: pyowm in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: geojson<2.4,>=2.3.0 in /usr/local/lib/python2.7/dist-packages (from pyowm)
Requirement already satisfied: requests<3,>=2.18.2 in /usr/local/lib/python2.7/dist-packages (from pyowm)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Requirement already satisfied: idna<3,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Requirement already satisfied: chardet<4,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests<3,>=2.18.2->pyowm)
Uninstalling mlbgame-2.5.0:
  Successfully uninstalled mlbgame-2.5.0
Collecting mlbgame from git+git://github.com/ajbowler/mlbgame.git@#egg=mlbgame
  Cloning git://github.com/ajbowler/mlbgame.git to /tmp/pip-build-6a9RCp/mlbgame
Requirement already satisfied: lxml in /usr/local/lib/python2.7/dist-packages (from mlbgame)
Installing collected packages: mlbgame
  Running setup.py install for mlbgame ... done
Successfully installed mlbgame-2.5.0
make: 'matrix/lib/librgbmatrix.a' is up to date.
If you didn't see any errors above, everything should be installed!

You'll need a config.json file to customize your settings. If you are updating
from an older version and you were required to run this install script again or
this is a fresh install, it's recommended we make a fresh one right now.
This will create a brand new 'config.json' file with default values so edit this
file with your own settings.

Would you like to do this now? [Y/n] y

You should now have a fresh config.json file you can customize with your own settings.

Installation complete! Play around with the examples in matrix/bindings/python/samples to make sure your matrix is working.
```

twfarley, 2020-08-27:
Can you post your config file?

t0xicchrome, 2020-08-27:
It is the one that is automatically generated. I have not made any modifications.

```
{
	"preferred": {
		"teams": ["Cubs"],
		"divisions": ["NL Central"]
	},
	"news_ticker": {
		"team_offday": true,
		"always_display": false,
		"preferred_teams": true,
		"traderumors": true,
		"mlb_news": true,
		"countdowns": true,
		"date": true,
		"date_format": "%A, %B %-d"
	},
	"standings": {
		"team_offday": false,
		"mlb_offday": true,
		"always_display": false
	},
	"rotation": {
		"enabled": true,
		"scroll_until_finished": true,
		"only_preferred": false,
		"rates": {
			"live": 15.0,
			"final": 15.0,
			"pregame": 15.0
		},
		"while_preferred_team_live": {
			"enabled": false,
			"during_inning_breaks": false
		}
	},
	"weather": {
		"apikey": "YOUR_API_KEY_HERE",
		"location": "Chicago,il,us",
		"metric_units": false
	},
	"time_format": "12h",
	"end_of_day": "00:00",
	"full_team_names": true,
	"scrolling_speed": 2,
	"debug": false,
	"demo_date": false
}
```

swemoney, 2020-08-27:
Config looks fine. Not sure why the specific error you're getting would be happening. I'm curious about why you're defining a 64x32 LED array in the command options but the script is detecting you have a 128x32.

t0xicchrome, 2020-08-27:
I am using a chain of 2 64x32 panels --led-chain=2

swemoney, 2020-08-27:
Ahhh. I missed the led-chain. Sorry about that. Would you be able to just completely delete the mlb-led-scoreboard directory and reclone/reinstall again?

swemoney, 2020-08-27:
also, it appears you're logged into your pi as root? if that's the case, you shouldn't need the `sudo`.

t0xicchrome, 2020-08-27:
I was copying the command from the readme. 

I have deleted the folder and redone the install.  The problem still persists
The error:

```
root@raspberrypi:~/mlb-led-scoreboard# python main.py --led-gpio-mapping=adafruit-hat-pwm --led-cols=64 --led-rows=32 --led-chain=2 --led-brightness=60
INFO (21:21:24): MLB LED Scoreboard - v4.0.0 (128x32)
Traceback (most recent call last):
  File "main.py", line 29, in <module>
    config = ScoreboardConfig("config", matrix.width, matrix.height)
  File "/root/mlb-led-scoreboard/data/scoreboard_config.py", line 22, in __init__
KeyError: 'preferred'
```

The config file:
```
{
	"preferred": {
		"teams": ["Cubs"],
		"divisions": ["NL Central"]
	},
	"news_ticker": {
		"team_offday": true,
		"always_display": false,
		"preferred_teams": true,
		"traderumors": true,
		"mlb_news": true,
		"countdowns": true,
		"date": true,
		"date_format": "%A, %B %-d"
	},
	"standings": {
		"team_offday": false,
		"mlb_offday": true,
		"always_display": false
	},
	"rotation": {
		"enabled": true,
		"scroll_until_finished": true,
		"only_preferred": false,
		"rates": {
			"live": 15.0,
			"final": 15.0,
			"pregame": 15.0
		},
		"while_preferred_team_live": {
			"enabled": false,
			"during_inning_breaks": false
		}
	},
	"weather": {
		"apikey": "YOUR_API_KEY_HERE",
		"location": "Chicago,il,us",
		"metric_units": false
	},
	"time_format": "12h",
	"end_of_day": "00:00",
	"full_team_names": true,
	"scrolling_speed": 2,
	"debug": false,
	"demo_date": false
}
```

Here is the install log:

```
root@raspberrypi:~/mlb-led-scoreboard# ./install.sh
Running rgbmatrix installation...
[... 147 lines cut ...]
Uninstalling mlbgame-2.5.0:
  Successfully uninstalled mlbgame-2.5.0
Collecting mlbgame from git+git://github.com/ajbowler/mlbgame.git@#egg=mlbgame
  Cloning git://github.com/ajbowler/mlbgame.git to /tmp/pip-build-Jsgm60/mlbgame
Requirement already satisfied: lxml in /usr/local/lib/python2.7/dist-packages (from mlbgame)
Installing collected packages: mlbgame
  Running setup.py install for mlbgame ... done
Successfully installed mlbgame-2.5.0
make: 'matrix/lib/librgbmatrix.a' is up to date.
If you didn't see any errors above, everything should be installed!

You'll need a config.json file to customize your settings. If you are updating
from an older version and you were required to run this install script again or
this is a fresh install, it's recommended we make a fresh one right now.
This will create a brand new 'config.json' file with default values so edit this
file with your own settings.

Would you like to do this now? [Y/n] y
rm: cannot remove 'config.json': No such file or directory

You should now have a fresh config.json file you can customize with your own settings.

Installation complete! Play around with the examples in matrix/bindings/python/samples to make sure your matrix is working.

```

swemoney, 2020-08-27:
Still not sure what would be causing it. Since most people don't ever log in as root, could you do a reinstall as the default user `pi` instead? So completely log out of the pi and log in again as just the regular old pi user.

t0xicchrome, 2020-08-27:
The problem was the /root directory permissions.  Even as root the program runs fine as long as it is not located within the /root directory. If I try to run it in the /root directory the error happens. Seems by default the root group does not have execution permissions in this directory. Setting the folder to 0750 allowed  Should have messed around with that. Thanks for the help.

closed 2020-08-27 by t0xicchrome