MLB-LED-Scoreboard/mlb-led-scoreboard #306: Add custom RSS feeds to news ticker?

plarsen007, 2020-10-04:
I have been using this project for 2 years and I absolutely love it.  I have a question that might seem a novice but is stumping me. I have been using the news ticker (offday) scoreboard (weather, mlb news, time) for the majority of the time and I would like to add different RSS feeds (local news etc….) what is the easiest way to do this? I have poked around headlines.py file thinking maybe id have luck creating another source but I’m doing something wrong. Is there a simple way to do this? Am I on the right track? 
Thank you!

swemoney, 2020-10-05:
Sadly, I didn't add the headlines feature with a lot of extendability in mind. Everything is in the `headlines.py` but there are a lot of different places you'd need to edit to add a new thing. This issue might be better as an enhancement request and I can try and rework it to be more extendable and customizable.

plarsen007, 2020-10-11:
Hey no worries! I really do like this project and if you ever get around to adding RSS feeds options that make this project even better! Thank you!

twfarley, 2022-04-08:
This isn't going to fall within scope for this project.

closed 2022-04-08 by twfarley