MLB-LED-Scoreboard/mlb-led-scoreboard #308: CircuitPython Support?

pjockey, 2021-01-18:
Not really an issue per-say, but wondering how much work it would be to make this run in CircuitPython? I have 4 64x32 boards and 2 Matrix Portal control boards. With this combination, I don't need the Pi nor the bonnet. I have no real programming skills but willing to work with anyone to try and bring to this to CircuitPython if feasible.

ty-porter, 2021-02-01:
The library that drives the LED matrices for this project is built specifically around Raspberry Pis. Refactoring would likely be a large undertaking to replace it with a CircuitPython driver (if there is one):

https://github.com/hzeller/rpi-rgb-led-matrix/tree/6668ff8b55fcfe5ea36cf22aa48155c9ced177a9

pjockey, 2021-02-02:
Thanks for the reply Tyler. Sorry to hear. I guess I will have to go and
purchase the bonnet.

scott

closed 2021-02-02 by ajbowler